letscontrolit/ESPEasy #114: Feature/change request: move hardwareInit above WifiConnect

dropax, 2017-02-27:
Especially with the sometimes unstable sonoffs it can take a long time until the configured boot states for GPIOs get applied. 
Found the reason in ESPEasy.ino (line 564ff): hardwareInit() is run after the wifi stuff is done. 

I use several sonoffs buried in lamps, they are powered through the original (wall)switches. This way the lamps can be used in the normal way. Just the delay between flipping the switch and the lamp coming on is irritating (especially if the wifi connection fails three times until AP mode is firered) and I think it would greatly improve the WifeAcceptanceFactor ;) to do the hardware init first. 

Is there technically anything speaking against moving hardwareInit() a few lines up?

psy0rz, 2017-02-27:
I will look into it to see what exactly happens in hardwareinit and what happens in the lines before it.

psy0rz, 2017-03-12:
fixed in v2.0.0-dev6. 

also after startup the sensor will be send immediately, instead of waiting arbitrary amounts of time (30 seconds )

closed 2017-03-12 by psy0rz