letscontrolit/ESPEasy #118: security issues in webserver.ino

psy0rz, 2017-03-01:
Password protection is not checked in some important handlers.

mvdbro, 2017-03-04:
Depending on which handler, this may be by design. Control handler is not protected by design. Password feature was implemented to protect the admin pages only. But of course things can be changed...
Changing could lead to unexpected errors for existing users that have set an admin password.

psy0rz, 2017-03-04:
I think handle_SDfilelist and handle_filelist?

now anyone can delete files without loggin in i think?

psy0rz, 2017-03-13:
i see, i think we should use update the documentation to state that it just protects the admin pages, but that ESPEasy is not ment to be in a public network.

beicnet, 2017-03-13:
and make all Password fields with ****** mask, please!

krikk, 2017-03-14:
the goal should be at least, that you need a password for everything where you can change things... even the cmd interface should need some sort of authorization token... even in a local "protected" network...

jroux1, 2017-04-17:
Adding my requirement. As per discussion on:
<ESP-host>/json - add security ?
https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=2962&p=15148#p15148

mvdbro, 2017-06-17:
Why should we need security on read only json output?
Some dev could try to enhance (or better redesign) the entire security handling, but this should not be done in V2.0.0.

psy0rz, 2017-06-19:
i agreed

SloMusti, 2017-07-01:
I believe there should be at least basic security on all user input, possibly implementing a read-only mode that is enabled/disabled by entering a password. I am using ESPEasy on private networks but must make sure no unauthorized playing with settings is possible.

OTA updates should require a password, because otherwise one can easily just upload a firmware that will brick the device. Could this maybe be done by simply having it in the Config page?

psy0rz, 2017-07-01:
@SloMusti we could but it also might break stuff. i agree we do security stuff in a later version and focus on getting a stable 2.0.0 version first.

SloMusti, 2017-07-01:
@psy0rz agreed on this

closed 2021-08-12 by TD-er