letscontrolit/ESPEasy #120: Forum Bug #16 OPEN: using single negative values in formulas produce wrong  values

psy0rz, 2017-03-01:


hphde, 2017-05-17:
I tried `-1*%value%` or `%value%*-1` to make the Wifi output positive but it seemed to have no effect. Is that covered by this issue?

uzi18, 2017-05-17:
No,  it is TaskValueSet with formula related.

psy0rz, 2017-05-18:
its probably the same underlying function, so yeah its covered :)

mvdbro, 2017-06-17:
No fix on this yet, but a workaround is to subtract from 0 to get the proper negative value.
So if you need -123.45, just type 0-123.45
To get RSSI as a positive value:
%value% * (0-1)

Grovkillen, 2018-04-08:
Still an issue?

hphde, 2018-04-17:
I'd vote for closing @Grovkillen

Grovkillen, 2018-10-23:
I tested it like this just now and this works just fine:
```
on rules#timer=1 do
taskValueSet,1,1,(-%syssec%)
publish,TEST,[Dummy#Test]
timerSet,1,1
endon
```
Please reopen if it's still an issue.

closed 2018-10-23 by Grovkillen