letscontrolit/ESPEasy #131: All Devices on single html page!

beicnet, 2017-03-02:
As the title says...

Can it be done? It's annoying to chase my Devices on 3 separated page per 4 Devices!

Thank you!

mvdbro, 2017-03-04:
It has been separated to avoid corrupted web pages due to insufficient RAM. This is still based on early core development where the entire web page had to be stored in a single string buffer first. I guess this approach can be changed, but it may take a complete webserver code overhaul to accomplish this.

beicnet, 2017-03-04:
@mvdbro Aha, okay, now I understand the reason of the separated HTML pages!

Thank you!

psy0rz, 2017-03-04:
ok, so maybe in a future version while we're doing a mayor overhaul anyway.

but not for now.

closed 2017-03-04 by psy0rz

beicnet, 2017-03-04:
@psy0rz Yes, that would be really good!