letscontrolit/ESPEasy #139: ESPEasy Mega MQTT support when not controller number 0

JustMe-NL, 2017-03-05:
Hi all,

I saw that the current Controller.ino has some hardcoded limits that although ESPEasy Mega supports up to 3 controllers, MQTT capable controllers are only supported if the are the first controller.

Now, there is a easy fix but should we bother? It just requires additional computational cycles and adds flash space to implement and we are still limited to only one MQTT capable controller. A better way would be to just mention the MQTT limitation in the wiki.

Why are our thoughts about this?
[Controller.ino.zip](https://github.com/letscontrolit/ESPEasy/files/819338/Controller.ino.zip)

psy0rz, 2017-03-07:
Yep, better yet: we should mention this in the webinterface. Or gray-out MQTT for the other controller numbers or something.

JustMe-NL, 2017-03-07:
I like the grayed out idea. Lets see if I can get that working.

psy0rz, 2017-03-07:
I dont now if we have that. Some day i would love to remove all the hardcoded html thats all over the place and create generic functions that create our html. (only for the default html stuff, custom html should stay possible offcourse)

closed 2017-03-08 by JustMe-NL