letscontrolit/ESPEasy #140: Build 148 fails with ESPCore Arduino 2.4.0 (ESP SDK 2.0.0)

diving91, 2017-03-06:
Hello,

Since I've issues with WiFi drops, I'm using the latest arduino core from [http://arduino.esp8266.com/nightly/package_esp8266com_index.json](url)

I can compile - I do not experience Wifi drops anymore, but I've an issue with Config/Advanced/Publish Template when trying to enter `device=%sysname%&taskid=%id%&cmd=%valname%&value=%value%`

only the part before the `&` is kept.
It sounds that some string parsing does not work with the latest ESP core and espressif SDK

I also tried to place this string in my .ino file and applied a reset command as described, but the same issue occurs

Any clue on how to fix this?

diving91

psy0rz, 2017-03-07:
Could you try building version 2 with Arduino 2.4.0?

You can just download the zip thats here: https://github.com/letscontrolit/ESPEasy/releases

Sources are in src (Rename it to ESPEasy)

Libraries are in lib. 

For the rest you can follow the default procedure that we've used in R148.

diving91, 2017-03-07:
Thanks for your reply.
I tried but I get lots of warning and error messages:
`Arduino: 1.8.1 (Windows 7), Board: "WeMos D1 R2 & mini, 160 MHz, 921600, 4M (1M SPIFFS)"

--skipped--
Compiling sketch...
D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_P044_P1WifiGateway.ino:17:0: warning: "BUFFER_SIZE" redefined [enabled by default]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_P020_Ser2Net.ino:10:0: note: this is the location of the previous definition

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino: In function 'boolean CPlugin_001(byte, EventStruct*, String&)':

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C001.ino:142:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C002.ino: In function 'boolean CPlugin_002(byte, EventStruct*, String&)':

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C002.ino:206:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C002.ino:206:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C002.ino:206:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C002.ino:206:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C002.ino:206:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C002.ino:206:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C002.ino:206:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C002.ino:206:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C002.ino:206:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C002.ino:206:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_C002.ino:206:150: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

[... cut ...]
D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_P044_P1WifiGateway.ino: In function 'bool checkDatagram(int)':

D:\Users\HervÃ©\Documents\Arduino\MySketches\ESPEasy\Source\ESPEasy\_P044_P1WifiGateway.ino:427:18: warning: unknown escape sequence: '\040' [enabled by default]

--skipped--
exit status 1
no matching function for call to 'SSD1306Wire::init(int&)'`

I get the same messages when building with master version of ESP core (2.3.0)
I tried (but failed) to remove some packages like the _P036_FrameOLED.ino
but then I get other errors

diving91, 2017-03-07:
News trial, I succeeded to compile by deleting the file _P036_FrameOLED.ino
in the configuration page: Controller Settings/Controller Publish, the same error as in version 148.
i.e. Publish template is truncated at the "&"

psy0rz, 2017-03-11:
Ok will try this later.

psy0rz, 2017-04-15:
can you try again with the latest git version?

diving91, 2017-04-15:
With the latest git version, it compiles perfectly (no errors, no warnings).
But still Controller Settings/Controller Publish in truncated at the "&"
(my publish template is device=%sysname%&taskid=%id%&cmd=%valname%&value=%value%)
I performed a factory reset in case of, but that does not solve the issue

diving91, 2017-04-15:
note that even with the default publish template for generic http controller it does not work:
the default is "demo.php?name=%sysname%&task=%tskname%&valuename=%valname%&value=%value%"
but when I hit to submit button it turns into
"demo.php?name=%sysname%"

psy0rz, 2017-04-15:
there was a seperate bug report for that righ

diving91, 2017-04-15:
what do you mean?

psy0rz, 2017-04-15:
oops sorry you're right

diving91, 2017-04-15:
No problem thank you

psy0rz, 2017-04-19:
i cannot find out where this is: 

> Config/Advanced/Publish Template when trying to enter device=%sysname%&taskid=%id%&cmd=%valname%&value=%value%
> only the part before the & is kept.

also the Generic http controller is storing these values correctly.

however i tried with platformio instead of Arduino IDE, so i still have to test that.

diving91, 2017-04-19:
Yes this is in the configuration of the generic http controller.
Indeed I'm not using platformio as the issue title says
I'm also trying to move to platformio but unfortunately I can't install it - see issue:
https://github.com/platformio/platformio-atom-ide/issues/376

psy0rz, 2017-04-20:
yeah and it should work in arduino ide as well. ill test it.

psy0rz, 2017-04-23:
i fixed some string handling issues in http generic. can you try agian with the latest git commit.

psy0rz, 2017-04-24:
its in release v2.0.0-dev8

diving91, 2017-04-24:
It works very fine by now - Many thanks for your effort, this is great!

closed 2017-04-24 by psy0rz