letscontrolit/ESPEasy #149: Replace "HTTP Generic" with "HTTP Generic Advanced" Controller

krikk, 2017-03-09:
As far as i see it the "HTTP Generic" Controller can be replaced with the "HTTP Generic Advanced" cause the advanced one does nearly everything the "basic" one does...

only feature of basic which is missing from the advanced is http basic authentication, but i will try to merge this to the advanced controller...  ...will try to get a pull request ready on the weekend :)

uzi18, 2017-03-09:
So for realy generic - template per sensortype feature is also needed.

psy0rz, 2017-03-09:
sounds good krikk:)

krikk, 2017-03-11:
TODO: an option to send all the values in one http request?

psy0rz, 2017-03-13:
that last pull request is pretty good. now we can send multiple values.

i did a bit of cleaning up and added it to testing.

a few notes for improvements: 

- the complex whileloops that delete unused variables should just call in a generic function, like this: void DeleteVarIndex(String &s, byte varindex).

In case of SENSOR_TYPE_SINGLE you can then just call:
 DeleteVarIndex(s, 2);
 DeleteVarIndex(s, 3);
 DeleteVarIndex(s, 4);

- handling of special stuff like SENSOR_TYPE_LONG should also be done in ReplaceTokenByValue(). That way we can remove the whole switch-statement in  "case CPLUGIN_PROTOCOL_SEND:" and simply call HTTPSend011().

after that there is more room for little improvements, making the code even smaller and better maintainable. :)

thanks for all the hard work!

uzi18, 2017-03-15:
OK, need to add also custom server key:
&key=SomeKey
and actual esp-ip-address:
&ip=1.1.1.1

for gpio have different template - will be possible to store it as different Controller?

Maybe it is good idea to hardcode some tamplates in HTTP  Generic Advenced?

krikk, 2017-03-17:
the key should be no problem you can add what ever text you want... the ip is also possible with %ip% after my newest pull requst is merged...

uzi18, 2017-03-17:
Ok and what about predefinied templates?

krikk, 2017-03-17:
in a first step i will be simplier to collect this templates on the wiki...

psy0rz, 2017-03-17:
indeed. later i want to make a piece of javascript that dynamicly loads the templates from github. so we can add templates on the fly without needing new and bigger firmware.

i'm doing a similar thing for my ledanim project. works great.

uzi18, 2017-03-17:
@psy0rz nice project :)

deejaybeam, 2017-04-06:
would be nice to add a "PATCH" method to HTTP Generic advanced

krikk, 2017-04-06:
PATCH method?

deejaybeam, 2017-04-06:
@krikk Yes! Currently supported is GET,POST,PUT and HEAD - to update variables in pimatic via restAPI PATCH is needed

closed 2017-04-06 by psy0rz