letscontrolit/ESPEasy #152: update wiki to explain flashing with the new FlashESP8266.exe in Windows

psy0rz, 2017-03-11:


beicnet, 2017-03-11:
@psy0rz +1 for that my friend! ;)

ShardanX, 2017-06-21:
May i suggest:
[]https://www.letscontrolit.com/wiki/index.php/Basics:_Connecting_and_flashing_the_ESP8266)

Please note the "Work in Progress"....
Suggestions and improvements welcome.

psy0rz, 2017-06-22:
cool looks pretty good!

psy0rz, 2017-06-22:
may i suggest to only make dangerous stuff red? stuff that might electrocute you or damagd hardware. now there is too much red: a few things should be bold instead. (like the hardware type suggestion or driver warnings)

if its done (or better what we currently have)  please link to it from the main page and perhaps from other pages and we can close this ticket.

ShardanX, 2017-06-25:
I've changed several "red" parts to bold.

Atm i'm discussing possibilities of structuring and archiving with grovkillen.
Too much orphaned but usefull parts in the wiki, many outdated stuff that should be archived & updated latest with the 2.0.0 release, 

Ideas welcome :)

psy0rz, 2017-06-25:
nice! thanks in advance guys :)

psy0rz, 2017-07-16:
how's this going? can we close the issue? :)

ShardanX, 2017-07-16:
The actual state is here:
https://www.letscontrolit.com/wiki/index.php/Basics:_Connecting_and_flashing_the_ESP8266

I'd like to add some screenshots of the flashing tool, but it's already usable i think

Grovkillen, 2017-07-17:
I will have time the next two days to fix the last bits.

Grovkillen, 2017-07-18:
@psy0rz @ShardanX I just added some screenshots and fixed some text. Please have a look.

Grovkillen, 2017-09-13:
@psy0rz Is there something else I can do or is this considered done?

psy0rz, 2017-09-14:
looks great! thanks :)

closed 2017-09-14 by psy0rz