letscontrolit/ESPEasy #160: Are those plugins from ESPEasyPluginPlayground still fully compatible with recent ESPEasy v2 builds?

beicnet, 2017-03-13:
As the title asks ;)

"Are those plugins from ESPEasyPluginPlayground still fully compatible with recent ESPEasy v2.0 builds?"

If not, can you include this few plugins for the next release?
1. [https://github.com/letscontrolit/ESPEasyPluginPlayground/blob/master/_P113_SI1145.ino](url)
2. [https://github.com/letscontrolit/ESPEasyPluginPlayground/blob/master/_P115_HeatpumpIR.ino](url)
3. [https://github.com/letscontrolit/ESPEasyPluginPlayground/blob/master/_P199_RF443_KaKu.ino](url)
4. [https://github.com/letscontrolit/ESPEasyPluginPlayground/blob/master/_P112_Power.ino](url)
5. [https://github.com/letscontrolit/ESPEasyPluginPlayground/blob/master/_P106_IRTX.ino](url)

Thank you!

psy0rz, 2017-03-13:
IRTX is already in v2.0. 

All plugins should be compatible with 2.0. The API stayed the same.

Could you add the other plugins yourself, rename them to a valid numbre and put a #ifdef PLUGIN_BUILD_DEV around them?

After that, test them and  send a pull request.

closed 2017-03-13 by psy0rz

beicnet, 2017-03-14:
@psy0rz Okay, thanks, I will try that!