letscontrolit/ESPEasy #164: BME280: Current configuration leads to a too high temperature

adb76, 2017-03-15:
Hello, 

the current configuration of the BME280 sensor is not meant to be used for a weather / climate measurement and therefore the chip is running to warm and shows a tempertature which is 1-1.5 degree to high. As described in the Bosch specification and also shown in the Adafruit advanced example [https://github.com/adafruit/Adafruit_BME280_Library/blob/master/examples/advancedsettings/advancedsettings.ino](url) the are at least three typical scenarios:

Weather / Climate Monitor:
```
bme.setSampling(Adafruit_BME280::MODE_FORCED, 
                    Adafruit_BME280::SAMPLING_X1, // temperature 
                    Adafruit_BME280::SAMPLING_X1, // pressure 
                    Adafruit_BME280::SAMPLING_X1, // humidity 
                    Adafruit_BME280::FILTER_OFF   ); 
                        
     // suggested rate is 1/60Hz (1m) 
     delayTime = 60000; // in milliseconds 
```

Indoor Navigation:

```
 bme.setSampling(Adafruit_BME280::MODE_NORMAL, 
                     Adafruit_BME280::SAMPLING_X2,  // temperature                                                              
                     Adafruit_BME280::SAMPLING_X16, // pressure 
                     Adafruit_BME280::SAMPLING_X1,  // humidity 
                     Adafruit_BME280::FILTER_X16, 
                     Adafruit_BME280::STANDBY_MS_0_5 );
``` 

Gaming:
```
bme.setSampling(Adafruit_BME280::MODE_NORMAL, 
                    Adafruit_BME280::SAMPLING_X1,   // temperature 
                   Adafruit_BME280::SAMPLING_X4,   // pressure 
                   Adafruit_BME280::SAMPLING_NONE, // humidity 
                  Adafruit_BME280::FILTER_X16, 
               Adafruit_BME280::STANDBY_MS_0_5 ); 
```

Since the typical weather / climate Scenario works in the "forced mode" a special call is necessary - here the example from Adafruit:
```
      // set to forced mode, i.e. "take next measurement" 
         write8(BME280_REGISTER_CONTROL, _measReg.get()); 
        // wait until measurement has been completed, otherwise we would read 
      // the values from the last measurement 
        while (read8(BME280_REGISTER_STATUS) & 0x08) 
    delay(1); 
```

I tested the different modes in a test environment and indeed with the weater / climate scenario the temperature was 1 degree lower compared to the other scenarios and thereofore closer to a SI7021 sensor.

So from my point of view it would be great to change the default to the weather / climate sensor scenario. Maybe it would be feasible to give the user the chance to select the appropriate scenario.

Kind regards.

krikk, 2017-03-15:
great suggestion, have one of these sensors in use by myself... if you can code it, submit it in form of a pull request, would be the best... if not i will take a look at it...

adb76, 2017-03-15:
So indeed I would be interested to create a pull request - but it would take some time. So if we are not in hurry ;-) ...

dzikus, 2017-03-16:
Please test my last changes commited in 0461f5bd96c5b3ab3e37c687b16bbb8d6da6acdf.

Profiles selected by user are best solution in my opinion.

adb76, 2017-03-16:
@dzikus: The change you commited is for the BMP280 not the BME280 right? For the BME280 from my point of view we should lower the default to sampling x1 for all three values (Pressure, Humidity, Temperature), and change to "Filter Off" and "Forced Measurement". That is what Bosch Sensortec recommends for a weather / climate usage. Unfortunately for the forced measurement we have to make a special call to the sensor each time. That is also where I want to look into.

dzikus, 2017-03-16:
Its for BME280 and for BMP280 also, they are very similar and have the same thermometer and barometer inside.

beicnet, 2017-03-16:
If I'm not wrong, then the BMP280 is same as BME280, the difference is that the BME280 didn't pass the humidity calibration and test process (actually it's a same sensor chip) and you got new product as BMP280.

psy0rz, 2017-03-23:
was this fixed by a pullrequest?

adb76, 2017-03-23:
Fixed with 1d1d69a13ff0ecc5422bb3b900107bb3013f389b.

closed 2017-03-23 by adb76

schwabe, 2018-04-16:
I basically have the same problem with both my BME680 and BME280 sensors. For the BME680 sensors I kind of expect this problem as they even have a heating element for air quality measurement (up to 12mA compared to max of 3,6µA of the BME280). 

Maybe adding a parameter to add a fixed offset might a stopgap/workaround issue here?

clowrey, 2018-12-06:
Just wanted to add that I thought I had a similar issue of temperature too high - but it was just me being silly and having the BME280 module soldered directly to the Wemos D1 Mini and the heat from the ESP8266 module was throwing off the temp readings... 

I moved the modules to 3" long wires and they are now two separate ESP8266 meters are quite happily reading within 2% on the humidity and .2 degree C difference when placed next to each other. 

They also agree very well with a raspberry pi meter I have based on the BME280 module.

Tom-Bom-badil, 2018-12-06:
@clowrey: Did you buy all those BME280 from the same vendor at the same time?

clowrey, 2018-12-06:
@Tom-Bom-badil yes they are all from the same vendor same batch. AliExpress...

Tom-Bom-badil, 2018-12-06:
@clowrey: Just wondering - did you ever compare the results with those of a calibrated, independent device?

TD-er, 2018-12-06:
This original issue is quite old.
In the mean time I have made some significant improvements on the plugin, which makes the reported temperature (and humidity) much more in line with other commercially available sensors.
The main issue the plugin for this sensor initially had, was the reading of the registers in the module.
It read them one register at a time, which may lead to inconsistent data. This would lead to temperature readings which were too high and thus humidity readings which are too low.

Apart from that, it is a rather small sensor which may easily heat up. So placing it at some distance of heat producing elements like the ESP module itself will also help improve the accuracy of the readings.

Tom-Bom-badil, 2018-12-06:
@TD-er: According to a calibrated humidty / temperature meter, the last batch of BME280 I bought (5 at once) connected to (also 5) ESPEasy devices were all were in a range of -0.79 ... +0.91 °C and +11.1 ... +19.2 rH. Reference point is the calibrated meter. Distance of sensor to D1mini is always ~20 cm. 

So: Quality of temperature is tolerable, but humidity is just a no-go. Just saying ..

/tom

kaefert, 2018-12-07:
@Tom-Bom-badil at what frequency are you reading them?

clowrey, 2018-12-07:
@Tom-Bom-badil I do not have a calibrated reference unfortunately.. it does seem to correlate at least within 5% RH to my dehumidifier readings.. I am in Hawaii with very high humidity as a baseline, even with dehumidifier the lowest I can get is 45% with reasonable energy usage on the dehumidifier.