letscontrolit/ESPEasy #165: Thingspeak issue with Build07

njordan77, 2017-03-16:
Hi, I'm having troubles with Build07 as Thingspeak does not receive any values anymore...in Build06 all was fine.

psy0rz, 2017-03-17:
which plugin you use for that?

njordan77, 2017-03-17:
whatever plugin is behind the "controller-1" settings....so simply add a controller via Thingspeak's IP address + APIkey....worked fine in v05, v06....now it v07 same settings but no sync of data towards portal...Regards

psy0rz, 2017-03-17:
strange, we didnt make that many changes between dev6 and dev7. 

can you downgrade to dev6 to make sure it still works in dev6?

krikk, 2017-03-17:
thingspeak enforces a 15 sec delay between any submit, have you set this...?

njordan77, 2017-03-17:
Thanks for the feedback....i know about the 15sec delay, which is set. But anyway it should work at least for 1 sensor....without this change. Did check again and again updated the APIkey...not it works - sorry. seems a copy-paste mistake...with a space that did not belong there. Thanks for the real quick support!

psy0rz, 2017-03-18:
ok thanks for the feedback.

closed 2017-03-18 by psy0rz