letscontrolit/ESPEasy #171: A few of issues...

beicnet, 2017-03-18:
**1.** Missing files from the zipped builds (empty folders):
**-** Adafruit_NeoPixel
**-** ArduinoJson
**-** IRremoteESP8266

**2.** Build version shown as 20000 instead maybe 200, no?

**3.** Node list not compatible with previous versions of builds, e.g. r146?
**-** r146 node list shows the Mega builds, but the Mega builds can't see or show the r146 node list

krikk, 2017-03-19:
@1: only the "source code" zip/tar.gz have this empty folders, the "ESPEasy_v2.0.0-dev7.zip" has working folders

beicnet, 2017-03-19:
@krikk I think, we need only one distribution ZIP file and to include all parts like, scr, lib, dist, bin,... to avoid further complications and misunderstanding. Anyway, inside the scr folder need to be another folder called "ESPEasy" contains the actual source/sketch files, because Arduino IDE wants to make folder and it only copies the ESPEasy.ino file nothing else.

psy0rz, 2017-03-23:
sorry we cant remove those .zip files. github adds them automaticly it seems. ill try if i can automaticly overwrite them with a proper one.

psy0rz, 2017-03-23:
buildversion is a problem: it should be a 16 bit integer, but there is no clean way to use that with the new versioning system.

currently we're at 2.0.0-dev7 so i used: 2 00 00

but this is not ideal since it doesnt use the 7.

any one has suggestions for a better system?

psy0rz, 2017-03-23:
also please open a new issue for every issue. having 4 issues in one is quite horrible for me ;)

psy0rz, 2017-03-23:
(the next time when you find new issues i mean)

beicnet, 2017-03-23:
@psy0rz No problem, next time I will open a new one for every issue...

beicnet, 2017-03-23:
@psy0rz I don't know about the build versioning system, but 204, 205, 206 and 207 are better option after (146, 147 and 148 revisions/versions).

beicnet, 2017-03-30:
@psy0rz Please read my last post in [https://github.com/letscontrolit/ESPEasy/issues/176](url), but, I can't fix that myself!

psy0rz, 2017-04-19:
fixed/reported in other  issues.

closed 2017-04-19 by psy0rz