letscontrolit/ESPEasy #175: Add support for IAQ Core C Sensor

nguyencong-khanh, 2017-03-21:
Add support for IAQ Core Sensor (CO2 & Tvoc). Please set value of I2C ClockStretchLimit parameter to 1000 at menu Tools --> Advanced.
[_P050_IAQCOREC.zip](https://github.com/letscontrolit/ESPEasy/files/858514/_P050_IAQCOREC.zip)

krikk, 2017-03-21:
why the change of i2c clock in the code, does it not work with the standard clock speed?

krikk, 2017-03-21:
and it would be nice if you could to this as git pull request, would speed up the integration into the codebase... ...btw really nice sensor, where did you buy it?

nguyencong-khanh, 2017-03-22:
@krikk , This sensors work only while set i2c clock to 1000, ( or other values, but i haven't tried yet). I brought it with LORA Dev Kit from AdvanWise in Taiwan.

krikk, 2017-03-22:
i think you mean the ClockStretchLimit , which should be 1000 for this sensor, because the clockspeed you set in you code is Wire.setClock(100000); (100khz) ...so will be great if you can do a pull request and document it on the wiki 👍

psy0rz, 2017-03-23:
indeed a pull request helps speed up development. (i merge most pull requests from my phone via the github page. if they look clean offcourse)

also dont forget to add the proper PLUGIN_BUILD_DEV ifdefs

Grovkillen, 2018-04-08:
Is this something that is going to be added as a PR? OP can give us a notice if this is still active and I'll open it again

closed 2018-04-08 by Grovkillen

njordan77, 2018-04-15:
hm, would also love to have this sensor available in main branch. is there any activity ongoing?

arrowcircle, 2018-09-21:
Any news on the support of this device?

sascha777, 2019-03-13:
Hi, I would also be interested in an IAQ support. 
Are there any news about this sensor?

pcace, 2022-03-06:
Any news on that topic?