letscontrolit/ESPEasy #176: Fix the CSS Style customization (extend)

beicnet, 2017-03-21:
Please add CSS Style Class for:

**1.** **Node list IP addresses**, like class="ipaddr"
**2.** Enabled sate in the **Controllers**, **Devices** and **Noticications** Tab, like class="enbl"
(btw...the Green and Red color for Yes and No it's not mandatory, those are hardcoded in the INO file)

Because of these two missing class, I can't fully customize my current ESPEasyMega style!

And also please FIX the orientation for the IDX/Value in the Devices Tab to be **CENTER** and for the Enabled state values to be centered.

Thank you! ;)

Some in-dev screenshots....

[image]

[image]

[image]

[image]

[image]

psy0rz, 2017-03-23:
could you make those changes,test them, and make a pull request?

matbe we should have a css-theme-directory with some nice example themes like yours.

psy0rz, 2017-03-23:
examples could go in a folder called dist/themes

beicnet, 2017-03-23:
@psy0rz I would really glad to do that, but I'm tooo tumb for using GitHub... :(
(Maybe I will try to edit the local INO file then to do some poll request)

krikk, 2017-03-28:
@beicnet you can do a very "simple" pull requst via the github page, just click the edit button, used this also for my first pull request :)
[image]

beicnet, 2017-03-28:
@krikk Okay! ;)

beicnet, 2017-03-29:
@krikk Can't fix it, it's not a CSS nor Style issue, I think it's a procedure issue putting 3x <BR> after the IDX/Variable value and it's beyond my programming knowledge, but I can lead you to it be able to fix it.

In the file WebServer.ino from line 1206 to 1212

```
for (byte controllerNr = 0; controllerNr < CONTROLLER_MAX; controllerNr++) {
 if (Settings.TaskDeviceID[controllerNr][x] != 0)
  reply += Settings.TaskDeviceID[controllerNr][x];
 if (x < CONTROLLER_MAX)
  reply += F("<BR>");
}
```

[image]

Thank you!

psy0rz, 2017-04-19:
this is fixed, right?

beicnet, 2017-04-19:
@psy0rz Not fixed nor corrected, I don't have that much programming skill to do it! :(

psy0rz, 2017-04-19:
didnt i merge a bunch of cosmetic pull requests of you? or was that other stuff :)

beicnet, 2017-04-19:
@psy0rz Yes, you did, but those was only "Naming" cosmetics stuff! :D

psy0rz, 2017-04-19:
ok:)

psy0rz, 2017-05-02:
@jkDesignDE could you look at these issues to while you're at it?

jkDesignDE, 2017-05-02:
OK - I will do it in the next days

psy0rz, 2017-05-13:
@beicnet with your and @jkDesignDE  you should be able to really pimp your ESPEasy to the max.

maybe put some screenshots/examples of your customisation on the wiki if you're done? :)

closed 2017-05-13 by psy0rz

beicnet, 2017-05-13:
@psy0rz Oki! ;)