letscontrolit/ESPEasy #183: How to move plugin from playground to espeasy?

tedenda, 2017-03-26:
Hello!

I want to know best practice how you move a plugin from playground to main ESPEasy branch.
In my case it is _P111_SenseAir.ino that I feel its ready to move to ESPEasy branch, it is tested for a long time and got a big update last week.

Best Regards
Daniel

beicnet, 2017-04-03:
@tedenda Just copy the desired INO file from the PlayGround into your ESPEasy directory and start again the ESPEasy.ino file.

tedenda, 2017-04-03:
No, I want the plugin to be moved from Playground  to main ESPEasy branch.

beicnet, 2017-04-03:
I don't know that! :( Sorry!

tedenda, 2017-04-03:
Maybe @psy0rz can give a good answer to this?

beicnet, 2017-04-03:
You can also ask @krikk too.

tedenda, 2017-04-03:
Thanks, good to know!

psy0rz, 2017-04-19:
tedenda, please rename the plugin to the ID number  and create a pull request. thats the way to go.

also please add #ifdef PLUGIN_BUILD_TESTING around the plugin.

closed 2017-04-19 by psy0rz