letscontrolit/ESPEasy #185: firmware.elf section `.text' will not fit in region `iram1_0_seg'

feiltom, 2017-03-27:
Hello 
when i build normal are OK but not test and dev it's normal ?
Environment normal_1024 [SUCCESS]
Environment normal_4096 [SUCCESS]
Environment test_1024   [ERROR] firmware.elf section `.text' will not fit in region `iram1_0_seg'
Environment test_4096   [ERROR] firmware.elf section `.text' will not fit in region `iram1_0_seg'
Environment dev_1024    [ERROR] firmware.elf section `.text' will not fit in region `iram1_0_seg'
Environment dev_4096    [ERROR] firmware.elf section `.text' will not fit in region `iram1_0_seg'

other question why include : 
#include <LiquidCrystal_I2C.h>
#include <Servo.h>
it's need only by plugin LiquidCrystal_I2C.h ex P012

I have dev a plugin for quirky nimbus but i need brzo_I2C (more speed and i need 400khz i2c) librarie and i need to disable some lib and plugin (ex LiquidCrystal_I2C ) without this my build generate also firmware.elf section `.text' will not fit in region `iram1_0_seg' in normal environment.

why not use brzo_I2C in espeasy ? it s more speed and seem less consumes ESP8266 ressource
ex :
http://www.electronics-lab.com/esp8266-ssd1306-oled-library-release/
ex : 
with and without brzo i2c implementation from a other project :
version with brzo implementation uses flash: 258 961 and ram: 34 248
version with wire.h implementation uses flash: 259 065 and ram: 34 392

Thanck you by advance
( sorry perhaps you prefer 3 issues ? )

krikk, 2017-03-28:
> Hello
when i build normal are OK but not test and dev it's normal ?
Environment normal_1024 [SUCCESS]
Environment normal_4096 [SUCCESS]
Environment test_1024 [ERROR] firmware.elf section .text' will not fit in regioniram1_0_seg'
Environment test_4096 [ERROR] firmware.elf section .text' will not fit in regioniram1_0_seg'
Environment dev_1024 [ERROR] firmware.elf section .text' will not fit in regioniram1_0_seg'
Environment dev_4096 [ERROR] firmware.elf section .text' will not fit in regioniram1_0_seg'

what are you using to compile? ...and what target platform do you have selected? never seen this error...

> other question why include :
#include <LiquidCrystal_I2C.h>
#include <Servo.h>
it's need only by plugin LiquidCrystal_I2C.h ex P012

LiquidCrystal_I2C.h is needed for plugin: _P012_LCD.ino
Servo.h is used for controlling servos see [here](https://www.letscontrolit.com/wiki/index.php/GPIO#Servo_motor_control)

> why not use brzo_I2C in espeasy ?

i did only a quick check, but as far as i see it, brzo_I2C is no drop-in-replacment for the currently used library, so we would have to re-write every plugin which uses i2c... and test all of this devices, this all just to save a few bytes... i think: to much work, for little outcome...

psy0rz, 2017-04-19:
we probably fixed this for now. let us know if its still a problem.

closed 2017-04-19 by psy0rz