letscontrolit/ESPEasy #188: Update IRremoteESP8266 library to canonical site.

crankyoldgit, 2017-03-29:
Hi,
  as one of the collaborators on the IRremoteESP8266 library, we've had a number of odd reports of our library not working coming from some ESPEasy users. I _think_ I've trace the root of the problem to your project referencing an unmaintained fork instead of the canonical version.

I'll create a pull request to fix what I can immediately see might be the issue, however, can you please check there is no other references your project has to the unmaintained version?

Thanks in advance.

psy0rz, 2017-04-01:
i updated it and trying to compile it against, suddenly all the versions give this error:

 /home/psy/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/test_1024/firmware.elf section `.text' will not fit in region `iram1_0_seg'

crankyoldgit, 2017-04-02:
The library isn't really big, but it has grown since the version you had referenced.

psy0rz, 2017-04-02:
it adds about 10k. i would have expected we would have way more rom left still. 

we probably hit some kind of limt. will have to find out how it works and perhaps write a script that lists the rom usage per plugin and go from there.

psy0rz, 2017-04-02:
btw thanks for informing us about that outdated repo :)

crankyoldgit, 2017-04-02:
Some quick googling, it appears it isn't the rom that is being exhausted, but the (i)ram.
http://www.esp8266.com/viewtopic.php?f=9&t=12754

Some of the interrupt routines for decoding incoming IR commands have been moved to iram for speed in the library. This hasn't been an issue for us, but given all the other libraries you use/include, you've probably exhausted the iram in the total build.

crankyoldgit, 2017-04-02:
We've got two interrupt procedures that have ICACHE_RAM_ATTR enabled, they are not particularly large. My guess is you've been close to the total iram budget for a while the minor recent changes there have pushed you over.

crankyoldgit, 2017-04-02:
FWIW, it seems it wasn't our library that was causing you issues. You already had iram problems prior to this. e.g. #185

psy0rz, 2017-04-02:
yep but that was only for that users environment and not reproducable. we also use travis to automaticly build every commit.

perhaps its related.

psy0rz, 2017-04-02:
yes indeed. i'm going to find out how to show the iram and other usage after compiling, and then write a script that automaticly determines usage for every plugin in our system. that we it will be easier to split of 'huge' plugins.

after that i'm going to find out what things we can do to reduce usage in general. 

until now we are just guessing about usage, instead of actually measuring it.

crankyoldgit, 2017-04-02:
Please ping me on that commit, I'm interested to see how you implement it.

psy0rz, 2017-04-02:
I found a tool that can show some more information:
```
psy@psypad ~/ESPEasy % ./memanalyzer.py /home/psy/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-objdump .pioenvs/dev_4096/firmware.elf
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8458|    1112
    rodata|           ReadOnly Data (RAM)|    3FFE8460|    3FFEACEC|   10380
       bss|      Uninitialized Data (RAM)|    3FFEACF0|    3FFF49C8|   40152
      text|            Cached Code (IRAM)|    40100000|    40107F00|   32512
irom0_text|           Uncached Code (SPI)|    40201010|    402827E3|  530387
Total Used RAM : 51644
Free RAM : 30276
Free IRam : 274
```

i found it here: https://raw.githubusercontent.com/SmingHub/Sming/develop/tools/memanalyzer.py

psy0rz, 2017-04-02:
this is with the infrared plugins disabled, as you can see we're already very limited.

i'll transform that pythonscript in something that tries to determine usage of all rams and roms per plugin.

crankyoldgit, 2017-04-02:
Yeah, IIUIC the iram limit is 32k, so you were/are right on it. I'm interested to see how much we use.
I think I've done everything we can at our end to reduce iram usage, but if you find something else, let us know.

crankyoldgit, 2017-04-02:
Hmm, we could split the library into a receive and a transmit portion. Only the receiver part needs iram, but that would break a lot of existing usage. Something for another day.

psy0rz, 2017-04-02:
yeah i'll let you know, there are probably many things we can do on our side first.

uzi18, 2017-04-02:
Try add option -C after -t to objdump

psy0rz, 2017-04-03:
The results are here: https://github.com/letscontrolit/ESPEasy/blob/mega/dist/Plugin_sizes.txt

Especially the IRAM usage is a problem currently. Without any plugins we only have around 2k left. Some plugins with ISR's need to use some IRAM. I'll see what i can do to reduce that first.

uzi18, 2017-04-03:
> mega/dist/Plugin_sizes.txt

Did You checked also without Controllers?

psy0rz, 2017-04-03:
yup

psy0rz, 2017-04-09:
@crankyoldgit using the IR receive functions uses 288 bytes of IRAM it seems. Thats probably because of the interrupt handler routine. Could you check if its easy to save some bytes in that function? Even 10s of bytes will help if its possible, without any compromises.

crankyoldgit, 2017-04-09:
@psy0rz I've been thinking of it ever since I saw your data. I haven't yet come up with an easy/good/safe way to do that yet. Now that I know how tight iram is, it's certainly a desired outcome. I'll try to let you know if I come up with something.

psy0rz, 2017-04-09:
Ok, if it cant be done, it cant be done. :)

Its too bad that we only have around 2k left to work with of the 32k. Makes you wonder what else can be done in the ESP8266/Arduino core libraries to reduce IRAM memory usage.

crankyoldgit, 2017-04-10:
Per the impending PR on our library, we should be reducing by about 28 bytes of IRAM when it gets updated.

I've got two other ideas to try that might reduce it further, but they are really horrible potential kludges.

psy0rz, 2017-04-10:
well if its kludgy dont do it:)

crankyoldgit, 2017-04-10:
That may be why I haven't done it. ;-)

crankyoldgit, 2017-04-29:
FYI, v1.2.0 of IRremoteESP8266 has been released with reduced IRAM usage.

closed 2017-06-21 by psy0rz

psy0rz, 2017-06-21:
thanks @crankyoldgit , the plugin that uses your lib went from 288 to 260 bytes IRAM. :)

very nice.

Edwin

commit 1c63589b (2017-06-21): upgraded IRremoteESP8266 from v1.0.2 to v1.2.0. fixes #188