letscontrolit/ESPEasy #191: Size of build too large to fit into 1MB

jbaumann, 2017-03-30:
Hi,

I've now tried to build the 8285 version. Both this and the 1024 version do not fit into 1MB. What could I strip down to get a version that I can use for this memory size?

Cheers, Joachim

krikk, 2017-03-30:
just delete plugins you do not need (everything with _P*), you can even delete unneeded controllers (_C*)....

jbaumann, 2017-03-31:
Ok, works. Thank you...

Should we have a special build target that e.g., uses a configuration file to decide which plugins should be included?

psy0rz, 2017-03-31:
yes we should, we already have defines for dev/test and normal plugins. 

we also should make a selection of 'minimal' plugins, that we want in the 8285 version.

krikk, 2017-03-31:
ideally we would have some sort of "wizard" which allows the user to select which plugins to include in the firmware... ...in a future far far away :)

psy0rz, 2017-04-03:
just wrote a memory analyser that shows us the memory usage per plugin (ram iram and rom).

krikk, 2017-04-03:
👍

psy0rz, 2017-04-03:
The results are here: https://github.com/letscontrolit/ESPEasy/blob/mega/dist/Plugin_sizes.txt

Especially the IRAM usage is a problem currently. Without any plugins we only have around 2k left. Some plugins with ISR's need to use some IRAM. I'll see what i can do to reduce that first.

jbaumann, 2017-04-03:
We could add a header file that defines a macro name for every plugin file, and have the plugin contents surrounded by an #ifdef. For instance for _P002_ADC.ino:

#include "enabled_plugins.h"
#ifdef _POO2_ADC
...
#endif

and in enabled_plugins.h:
// #define _P001_Switch
#define _POO2_ADC
...

Then we could use a program (a wizard) that takes the information from Plugin_sizes.txt and displays a menu choice to the user. When we additionally add the dependencies then this could work like a charm (or e.g., a bit like the linux kernel configuration)...

And fantastic work. I still have to understand the different types of RAM and their limits.

uzi18, 2017-04-03:
> mega/dist/Plugin_sizes.txt

Try also without plugins with different esp8266 versions... realy curious
if there is best one.

beicnet, 2017-04-03:
I don't know guys, but ESPEasy was not been developed for led controlling, thus I think the whole NeoPixel and maybe the Motor Control is not suitable for it (maybe better to remove it), there are more simple and cool sensors out there to deal with, don't you think so?!

krikk, 2017-04-04:
the main problem i see with "customized" build like @jbaumann describes it, is how do we get a "deploy able" build environment to the user?

jbaumann, 2017-04-04:
Wouldn't it be three steps?
1. Implement a mechanism that allows to turn on and off the build of specific plugins.
2. Create a wizard for this mechanism that can be used when building on your own computer.
3. Create a web application (like https://nodemcu-build.com/) that allows to build a customized version for people who do not want to build themselves.

The complexity of using the mechanism drops with each step...

krikk, 2017-04-04:
@psy0rz how do i run this memory analyzer?? ...you are developing on a *nix like system? ...is it possible the path things are wrong on a windows system?

psy0rz, 2017-04-04:
currently it needs the git and platformio commandline programs. 

i just hacked it together quickly without worrying about other operating systems too much.

in linux you have to start it with the path to the correct objdump tool that is part of the platformio toolkit:

```
$ ./memanalyzer.py /home/psy/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-objdump 
```

to make things easier i can make it a part of travis, so that you can watch the travis output for the results. (you would have to do a pull request first)

psy0rz, 2017-04-04:
ill try to get a buildservice like nodemcu-build.com running.

jbaumann, 2017-04-08:
Hi,

I have written a small wizard in Java, but I would need the max values for 1M and 4M configurations (assuming the 8285 is the same as a 8266 with 1M). @psy0rz could you provide me with the values for the different categories listed in the output from your memanalyzer?

Cheers, Jaochim

jbaumann, 2017-04-11:
Hi,
if you want to play with the program I can either send it to you or append here. 

Later on, when I have time, I will put it on Github. @psy0rz, I still need the correct values for the memory limits...

Cheers, Joachim
PS: It now produces a correct header file, examines the plugin files in the changeable source directory, is configurable using a YAML file, highlights the changes etc.

psy0rz, 2017-04-19:
We're now also building firmware for the 8285 by default. (called normal_8285.bin )

So this bug is fixed, correct?

closed 2017-04-19 by psy0rz

jbaumann, 2017-04-19:
Yes, I think so.