letscontrolit/ESPEasy #196: Web interface bug, need to be fixed!

beicnet, 2017-04-03:
I think it's a procedure issue putting 3x `<BR>` after the IDX/Variable **value** and it's beyond my programming knowledge, but I can lead you to it to be able to fix it.

In the file WebServer.ino from line 1206 to 1212

```
for (byte controllerNr = 0; controllerNr < CONTROLLER_MAX; controllerNr++) {
 if (Settings.TaskDeviceID[controllerNr][x] != 0)
  reply += Settings.TaskDeviceID[controllerNr][x];
 if (x < CONTROLLER_MAX)
  reply += F("<BR>");
}
```

[image]

Thank you!

psy0rz, 2017-04-12:
Thats not a bug, thats just a cosmetic cleanup so things look nicer.

beicnet, 2017-04-12:
@psy0rz But, can you fix it for me?! ;)

psy0rz, 2017-05-02:
if you're lucky this is fixed in the last commit, and otherwise @jkDesignDE  might fix it. :)

beicnet, 2017-05-02:
Yeah, let's hope so, that @jkDesignDE really good!

commit 9d5afd1e (2017-05-03): Fixed #196 and rework of IDX display
closed 2017-05-03 by psy0rz