letscontrolit/ESPEasy #197: About fit into 1MB...

beicnet, 2017-04-03:
Hi there,

I don't know guys, but ESPEasy was not been developed for years for led array controlling, thus I think the whole NeoPixel and the Motor Control is not suitable for it (maybe better to remove it, maybe only 2% of ESPEasy users use it or maybe even less), but, there are more simple and usefully, cheap sensors out there (what I already have and I can't use them) to deal with like (LM75A, SHT30, HTU21D, DHT21, AM2320, TEMT6000, SI1145 - to be officially included), what do you think of it?!

Just my opinion! ;)

krikk, 2017-04-04:
i do not know what ESPEasy was developed for and it does not really matter, because i think we should include as much functionality as we can do, so instead of removing things i strongly opt for "customizing" builds, so that every user can select the plugins which he needs. even if a trade-off of this would be that we could no longer have a all-in-one build because it would no longer fit because of iram limits...

btw. HTU21D is already supported, because it is a drop-in replacment for si7021

beicnet, 2017-04-04:
@krikk  Yes, that would be good if we can choose what to include into the custom build!

Also, I know for the SI7021, because I have them too, but other users not.

So, I prefer for marking like this:

Temperature & Humidity - SI7021/HTU21D

Anyway, yesterday I tried to make a pull request and the build failed, I just added a back slash "/".

Temperature & Humidity - BMP085/BMP180

Also I prefer for the "Temperature & Humidity" to be replaced for "Environment", (**less char space**).

Like this:

Environment - SI7021/HTU21D
Environment - BMP085/BMP180
Environment - BMP280
Environment - BME280
Environment - DHT11/12/22
Environment - DHT12 (i2c)
Environment - SHT1x
Environment - DS18B20
Environment - MS5611

etc...

psy0rz, 2017-04-04:
I'll try and see if we can get a build service running like nodemcu-build.com

alexschwantes, 2017-04-09:
That sounds awesome, I have been thinking that would be a great addition for a while now. Maybe they would consider sharing the code to make things easier?

psy0rz, 2017-04-19:
duplicate of #191

closed 2017-04-19 by psy0rz