letscontrolit/ESPEasy #198: MFRC-522 RFID Module?

beicnet, 2017-04-03:
Hi there,

And what about MFRC-522 RFID Module (Blue) implementation?

Cheaper than any other already implemented RFID Module by two times!

Kind regards,
Viktor

patrickse, 2017-11-29:
👍

uzi18, 2017-11-29:
But has got very short range

vikasjee, 2017-12-11:
Do you plan to support this?

beicnet, 2017-12-11:
Specification

- Module Name:MF522-ED
- Working current：13 - 26mA / DC 3.3V
- Standby current：10 - 13mA / DC 3.3V
- Sleep current：<80uA
- Peak current：<30mA
- Working frequency：13.56MHz
- **Card reading distance ：0～60mm** （Mifare1 card）
- Protocol：SPI
- Data communication speed：10Mbit/s Max.
- Card types supported: Mifare1  S50, Mifare1 S70, Mifare UltraLight, Mifare Pro, Mifare Desfire
- Dimension：40mm × 60mm
- Working temperature：-20—80 degree
- Storage temperature：-40—85 degree
- Humidity：relevant humidity 5%—95%

tuxmartin, 2018-03-08:
Please add support of RC522.

I found it on wishlist https://www.letscontrolit.com/forum/viewtopic.php?t=736
Do you have a time estimate?

Thank you!

Grovkillen, 2018-04-08:
Any developer that got this unit and its planning to have it implemented?

TD-er, 2018-04-08:
Yep, it's here on the big pile of stuff to implement.

Grovkillen, 2018-04-08:
Cool, I will order one as well... No hurry though! :smiley:

Tueftler1983, 2018-04-22:
is there a Plan or date, how did it supportet?? i use at this time a extra nodemcu to integrate it in Fhem

TD-er, 2018-04-22:
At this moment all available dev time is put into making stuff stable again.
There are currently too many issues:

- wifi
- webpage loading
- damaged settings
- all kinds of other issues leading to instability.

jensfr1, 2018-07-05:
esp-rfid works just fine with the rfid reader. 

https://github.com/omersiar/esp-rfid/releases

plin2, 2019-08-11:
Well, what's the state one year later? I have ESPeasy running on an ESP32 with OLED display. The only thing missing is my RC522 RFID reader.

TD-er, 2019-08-11:
@plin2 
I have not done anything with this plugin.
The reader may have moved from the "big pile of stuff to implement" into a nice box labeled "Sensors for ESPeasy".

I have not had the time yet to have a good look at it.

uzi18, 2019-08-11:
@TD-er my reader have range about 1-2cm

s0170071, 2019-11-15:
I found some code on the internet and scetched a plugin. It just reads the RFID serial number which is okay for my door lock. The rest I do in OpenHab anyway.  The RFID is pushed to the (MQTT) controller the very moment the card is read.

Use: 
1. copy code to an empty .ino file, put it in the ESPEasy folder. (ArduinoIDE)
2. get the MFRC522 library. I used the latest version, 1.4.5. Just unzip it to your libraries folder. 
3. compile...

```
/*
* Typical pin layout used:
* -----------------------------------------------------------------------------------------
*             MFRC522      Arduino       Arduino   Arduino    Arduino          Arduino      WemosD1 /ESP8266
*             Reader/PCD   Uno/101       Mega      Nano v3    Leonardo/Micro   Pro Micro
* Signal      Pin          Pin           Pin       Pin        Pin              Pin
* -----------------------------------------------------------------------------------------
* RST/Reset   RST          9             5         D9         RESET/ICSP-5     RST          D0/GPIO16
* SPI SS      SDA(SS)      10            53        D10        10               10           D8/GPIO15
* SPI MOSI    MOSI         11 / ICSP-4   51        D11        ICSP-4           16           D7/GPIO13
* SPI MISO    MISO         12 / ICSP-1   50        D12        ICSP-1           14           D6/GPIO12
* SPI SCK     SCK          13 / ICSP-3   52        D13        ICSP-3           15           D5/GPIO14
*/
#include <SPI.h>
#include <MFRC522.h>
#define RST_PIN   D0     // SPI Reset Pin, optional
#define SS_PIN    D8    // SPI Slave Select Pin
//#######################################################################################################

#define PLUGIN_017
#define PLUGIN_ID_017         17
#define PLUGIN_NAME_017       "RFID - RC522"
#define PLUGIN_VALUENAME1_017 "Tag"
MFRC522 *mfrc522 = nullptr;
boolean Plugin_017(byte function, struct EventStruct *event, String& string)
{
  boolean success = false;
  switch (function)
  {
    case PLUGIN_DEVICE_ADD:
      {
        Device[++deviceCount].Number = PLUGIN_ID_017;
        Device[deviceCount].Type = DEVICE_TYPE_SINGLE;
        Device[deviceCount].VType = SENSOR_TYPE_LONG;
        Device[deviceCount].Ports = 0;
        Device[deviceCount].PullUpOption = false;
        Device[deviceCount].InverseLogicOption = false;
        Device[deviceCount].ValueCount = 1;
        Device[deviceCount].SendDataOption = true;
        Device[deviceCount].TimerOption = false;
        Device[deviceCount].GlobalSyncOption = true;
        break;
      }
    case PLUGIN_GET_DEVICENAME:
      {
        string = F(PLUGIN_NAME_017);
        break;
      }
 
    case PLUGIN_GET_DEVICEVALUENAMES:
      {
        strcpy_P(ExtraTaskSettings.TaskDeviceValueNames[0], PSTR(PLUGIN_VALUENAME1_017));
        break;
      }
 
    case PLUGIN_WEBFORM_LOAD:
      {
         // addFormPinSelect(F("Reset Pin"), F("taskdevicepin3"), CONFIG_PIN3);
        success = true;
        break;
      }
 
    case PLUGIN_INIT:
      {
        if (mfrc522 == nullptr){
        mfrc522 = new MFRC522 (SS_PIN, RST_PIN);   // Instanz des MFRC522 erzeugen
        //SPI.begin();         // Initialisiere SPI Kommunikation
        mfrc522->PCD_Init();  // Initialisiere MFRC522 Lesemodul
        }
        break;
     
      }
 
    case PLUGIN_TEN_PER_SECOND:
      {
        static unsigned long tempcounter = 0;
        static byte counter;
        static byte errorCount=0;
 
        counter++;
        if (counter == 3 )
        {
        counter=0;
        if (mfrc522->PICC_IsNewCardPresent() && mfrc522->PICC_ReadCardSerial() ) {
          String log = F("PN532: Tag: ");
         
          // put only the first 4 bytes to user variables 0 and 1
          unsigned long key =  mfrc522->uid.uidByte[0];
          for (uint8_t i = 1; i < 4; i++) {
            key <<= 8;
            key +=  mfrc522->uid.uidByte[i];
          }
          log+=key;
          UserVar[event->BaseVarIndex] = (key & 0xFFFF);
          UserVar[event->BaseVarIndex + 1] = ((key >> 16) & 0xFFFF);
      
          // switch off card to be able to look for others
          mfrc522->PICC_HaltA();
          addLog(LOG_LEVEL_INFO, log);
          sendData(event);
          }
        }
        break;
      }
  }
  return success;
}
```

TD-er, 2019-11-15:
@s0170071 If it is useful, can you make a PR for it with the proper plugin ID set?

vincnico, 2019-11-18:
@s0170071
Hi , seems interesting, but I would like to be sure to understand => is this allow to run a MFRC522 with a NodeMcu and EspEasy ?  and so enable the configuration on espeasy web interface ?
I am using Jeedom + some NodeMCU with espeasy. 
I am lookink for a solution with espeasy to have my RC522 rfid reader connected to Jeedom

thank you

pascalsaul, 2020-08-02:
> @s0170071
> Hi , seems interesting, but I would like to be sure to understand => is this allow to run a MFRC522 with a NodeMcu and EspEasy ? and so enable the configuration on espeasy web interface ? [...]

Yes @vincnico , I just tested it as a custom plug-in with little modifications and works with the latest MFRC522, NodeMCU and latest build from ESPEasy_ESP82xx_mega-20200721. Only question I've if the tag should be outputted as decimal or hex value... I'm not sure what's the default in other integrations?

Kev0123, 2021-02-05:
> > @s0170071
> > Hi , seems interesting, but I would like to be sure to understand => is this allow to run a MFRC522 with a NodeMcu and EspEasy ? and so enable the configuration on espeasy web interface ?
> > I am using Jeedom + some NodeMCU with espeasy.
> > I am lookink for a solution with espeasy to have my RC522 rfid reader connected to Jeedom
> > thank you
> 
> Yes @vincnico , I just build it as a custom plugin with some modifications and works with the latest MFRC522 library, NodeMCU v1 and latest available build: ESPEasy_ESP82xx_mega-20200721.
> 
> It only accepts MIFARE classic cards and ignores mobile phones with NFC enabled or Credit Cards. They generate random numbers every time so are useless to use since you can't check them. In the logging I added also debug information and under devices (and MQTT) the decimal value is posted to use with your integrations.
> 
> Logging:
> 
> ```
> RFID-RC522: Type: MIFARE 1KB, Hex: 050DEFD0700000, Dec: 84799440
> RFID-RC522: Your tag is not of type MIFARE Classic but PICC compliant with ISO/IEC 14443-4
> RFID-RC522: Type: MIFARE 1KB, Hex: 3985AFB0, Dec: 965062576
> ```
> 
> MQTT:
> 
> ```
> esp02/rfid/tag 84799440
> esp02/rfid/tag 965062576
> ```
> 
> Within ESPEasy I can't get the hex value being returned so I stick with the decimal number which is also fine. I tried something with the "SENSOR_TYPE_STRING" but then I get float conversion error with "UserVar[event->BaseVarIndex]".
> 
> I'm personally also looking at: https://www.youtube.com/watch?v=ENMul9eAB00 (https://github.com/esprfid/esp-rfid)
> 
> @TD-er I've used 99 as the plugin number which is free. I think I can make a pull request if somebody can test this plugin also.
> 
> Compiled for [ESP_Easy_mega_20200802_normal_ESP8266_4M1M](https://www.saul.nu/downloads/RFID-RC522.bin)

I'm searching for this solution for a long time. Sadly I cant get your compiled plugin to work. I don't get any output. Could you compile it with the latest release? 

Regards

uzi18, 2021-02-05:
@Kev0123 If you got problem with configuration please attach some photos/screenshots

Kev0123, 2021-02-05:
@uzi18 if I wire it according to the sheet above I get absolutely ni readout.
[image]

The only config is the GPIO.

s0170071, 2021-02-05:
pls check your wiring.  pin numbers are at the top of the code.  and 
remember to enable the spi in the hardware settings tab.

tonhuisman, 2021-02-05:
And please don't use a dash in the task name as that will cause issues with rules processing. (Not all builds have the validation  active, but it is causing issues when dashes are used).

Kev0123, 2021-02-05:
The wiring should be according your layout (by far not the best soldering, only temp setup ;-))
[image]

SPI was indeed not enabled 🤦 
But now after a reset my webserver is unreachable

uzi18, 2021-02-05:
So you have connection problem somewhere

Kev0123, 2021-02-05:
If I try an arduino scetch with the same setup its working fine. I just reflashed espeasy and now the webserver is available again. There is still no output for the plugin

uzi18, 2021-02-05:
please attach photo from top

Kev0123, 2021-02-05:
[image]

Kev0123, 2021-02-05:
Okay, I dont know why, but now I get a readout. Tomorrow Ill test with different hardware, maybe the esp is broken. I got continous reconnects just like with my own arduino sketch.

TD-er, 2021-02-05:
Check the 3v3 on the ESP to see if it is stable.
Also the RFid close to the ESP WiFi antenna is maybe not a good idea.

Kev0123, 2021-02-06:
After attaching longer wires my problem is solved. You can make a PR in my oppinion. 
Would be cool if there would be a variable "card present". For my task I could play and stop music easily this way.
[image]

Thank you all for the help!

pascalsaul, 2021-02-06:
https://www.saul.nu/downloads/RFID-RC522-20210114.bin

I've compiled and fixed a new version of the plugin with the latest mega-20210114 and rfid 1.4.8 release for anyone who is interested. If you upgrade you should choose to a new plugin since the plugin number was overlapping.

twinbee77, 2021-02-07:
I have try the version from pascalsaul
But i dont have a readout then I checked the wiring several times.nothing happend
then i try to the plugin from ESPEasyPluginPlayground.the same result=nothing.
then i try the snippet from s0170071. Nothing
When i used the example sketch from the ardunio ide all works fine.
someone has an idea where I can still look for mistakes

s0170071, 2021-02-07:
spi is enabled in the hardware tab?

twinbee77, 2021-02-07:
I compiled your code with latest esp-mega version.
It runs  now.i don't know where my mistakes were.
when i hold the card on the reader it shows a number always even whn i take away the card.
how can I achieve that it jumps back to 0 after removing the card?

Kev0123, 2021-02-07:
> I compiled your code with latest esp-mega version.
> It runs now.i don't know where my mistakes were.
> when i hold the card on the reader it shows a number always even whn i take away the card.
> how can I achieve that it jumps back to 0 after removing the card?

That would be the desired behaviour like I wanted.
This would make it possible to detect a present card and switch states according this. So thumbs up from me as well for this idea!

TD-er, 2021-02-07:
That has been implemented for other RFid plugins, by @uzi18 and @tonhuisman 
Can one of you take a look at this one too?

tonhuisman, 2021-02-07:
Can I safely assume that [P129](https://github.com/letscontrolit/ESPEasyPluginPlayground/blob/master/_P129_RC522_RFID.ino) in the PluginPlayground is the latest source for this plugin? If that's the case, then I can add that feature, like I did for other RFID plugins.
Don't have this hardware btw, I ordered a PN532 because a) it supports I2C, and b) does accept more cards/tags (it's less picky it seems). And that makes the price less of an issue 😃 (AliX: $1.41 - $2.91 for the PN532 vs $1.13 - $1.43 for the RC522, excluding shipping).

twinbee77, 2021-02-07:
The p129 plugin in the PluginPlayground dont run for me.i have several times compiled but no success.
the only plugin that works is this from s0170071 or the compiled version from pascalsaul.
My programming skills are very poor.Somewhere in the code must be placed a command to write a zero after take away the card.
I think somewhere between this part of the code:

 case PLUGIN_TEN_PER_SECOND:
      {
        static unsigned long tempcounter = 0;
        static byte counter;
        static byte errorCount=0;
 
        counter++;
        if (counter == 3 )
        {
        counter=0;
        if (mfrc522->PICC_IsNewCardPresent() && mfrc522->PICC_ReadCardSerial() ) {
          String log = F("PN532: Tag: ");
         
          // put only the first 4 bytes to user variables 0 and 1
          unsigned long key =  mfrc522->uid.uidByte[0];
          for (uint8_t i = 1; i < 4; i++) {
            key <<= 8;
            key +=  mfrc522->uid.uidByte[i];
          }
          log+=key;
          UserVar[event->BaseVarIndex] = (key & 0xFFFF);
          UserVar[event->BaseVarIndex + 1] = ((key >> 16) & 0xFFFF);
      
          // switch off card to be able to look for others
          mfrc522->PICC_HaltA();
          addLog(LOG_LEVEL_INFO, log);
          sendData(event);
          }
        }
        break;
      }
  }
  return success;
}

tonhuisman, 2021-02-07:
I'll have a look, your adaptions seem pretty OK, except you have hard-coded the GPIO pins, I'll try to fix that too.

TD-er, 2021-02-07:
@tonhuisman 
For SPI you cannot set a lot of pins, as long as you make a proper selection between ESP32 and ESP8266.

Kev0123, 2021-02-07:
May someone could upload a compiled .bin when the "zero-readout" is working? I can't even get this to compile

tonhuisman, 2021-02-07:
> I can't even get this to compile

Working on it... it's compiling here 😉 

Question (for all users of this plugin):
AFAICS, currently the plugin repeats sending the last tag read until a new tag is detected, is that the desired behavior, or should it only send a tag once? (to make it consistent with the other RFID plugins)

Kev0123, 2021-02-07:
For me It would be desired if it would only send once when the tag is placed. Otherwise the attached script would be triggered continously

tonhuisman, 2021-02-07:
That is how I'm crafting it now, if only to make it work in the same way as the other RFID plugins currently available.

twinbee77, 2021-02-07:
only once when the tag is placed.
and when no card is placed thy should send 0

tonhuisman, 2021-02-07:
I've put in the requested changes, and started preparations for a PR, so I assigned a new plugin number: Plugin 111. This implies that you will have to re-add the plugin and set any configuration as required. (More work is planned to enable multiple instances of the plugin and somewhat lower memory use when not used.)
Option 'Automatic Tag removal' is On by default, and 'Event on Tag removal' option is Off by default. (that's similar to the Wiegand RFID and PN532 RFID plugins)

A screenshot of the settings:

[image]

Created .bin files for ESP8266 (normal, 4M) and ESP32 (test, 4M).
[ESPEasy_20210207_normal_ESP8266_4M1M_test_ESP32_4M316k-MFRC522.zip](https://github.com/letscontrolit/ESPEasy/files/5939127/ESPEasy_20210207_normal_ESP8266_4M1M_test_ESP32_4M316k-MFRC522.zip)

twinbee77, 2021-02-07:
I give it a try

twinbee77, 2021-02-07:
I think you are my hero today
When i hold the card they show me the tag after 500 ms it goes back to 0.
sending to domoticz is possible it shows correct after i mark "Event on Tag removal:"
now the plugin work how it should
many thanks.
p.s.
can i have the .ino file then i can compile my self

pascalsaul, 2021-02-07:
[_P144_MRC522.txt](https://github.com/letscontrolit/ESPEasy/files/5939229/_P144_MRC522.txt)
 @tonhuisman I added also some enhancements so hereby my code..

Kev0123, 2021-02-07:
Am I wrong or is the no actual removal detection? It just times out the 500ms and the reset to the desired value?
Is the a way to actuallyreset when the card is removed from the reader?

s0170071, 2021-02-07:
has anyone looked at the library? maybe there is a method like cardPresent() ?

tonhuisman, 2021-02-07:
> 
> 
> [_P144_MRC522.txt](https://github.com/letscontrolit/ESPEasy/files/5939229/_P144_MRC522.txt)
> @tonhuisman I added also some enhancements so hereby my code..

Thank you.
The main difference I see is that you filter on the type of card (MIFARE_MINI, MIFARE_1K, MIFARE_4K). Why are the other card types not supported? Is the returned value too long (to be stored in a `long`)? Current code just returns the lowest 4 bytes as a `long`, for every type of card.

> Am I wrong or is the no actual removal detection?

You are correct. The code is not interrupt driven (and I have no desire to open that can of worms). That is why the removal time is configurable. (250 msec to 60000 msec).
I'll think of how I can add that feature, if the driver actually supports it.

> maybe there is a method like cardPresent() ?

Well, there is `IsNewCardPresent()`, and that is used to detect a new card (same card is ignored, it seems)
Alternative could be to allow a 'delay' of 0, so it is reported as removed ASAP (in fact 100 msec, as that's the scan frequency, 10 times per second, next faster interval would be 50 times per second, but I don't think 20 msec (including some overhead) is long enough to properly read the uid from all card types).

Kev0123, 2021-02-07:
> I'll think of how I can add that feature, if the driver actually supports it. The driver does not indicate removal of a card, you just ask it 'is there a card', and if it then finds one, it reads the uid and you can fetch that.
As I wrote some time ago this project has the detection implemented, only that the reconnection error to mqtt is persistent.
Maybe this could help ?
[https://haus-automatisierung.com/projekt/2018/12/16/projekt-playbox-mqtt-fhem-spotify.html](url)

tonhuisman, 2021-02-07:
> this project has the detection implemented,

It uses blocking code, and as that's no problem for Arduino, it is for ESPEasy, so not a feasible solution.

I think the alternative of allowing a reset time of 0 (in fact ASAP) is the way to go, and with Event on Tag removal enabled, MQTT events should be happy too.

Kev0123, 2021-02-07:
> I think the alternative of allowing a reset time of 0 (in fact ASAP) is the way to go, and with Event on Tag removal enabled, MQTT events should be happy too.

So the behaviour would be:
place card-> MQTT receives ID -> instant reset to 0 -> MQTT receives 0
remove card-> nothing happens at all ?

twinbee77, 2021-02-07:
Wrong plugin file for self compile.
It looks not like the precompiled version.
When i compile this it looks 
[image]

tonhuisman, 2021-02-07:
> So the behaviour would be:
> place card-> MQTT receives ID -> instant reset to 0 -> MQTT receives 0
> remove card-> nothing happens at all ?

Not sure, as I said, I don't have the (any) RFID hardware (yet), so haven't been able to test myself.

> Wrong plugin file for self compile.

I have only published a few .bin files, no updated source code yet, I'll make a PR for it, but not sure if I can accomplish that today. (There also is life, and work on weekdays 😆)

twinbee77, 2021-02-07:
Ok im waiting i test the .bin files meanwhile.
thanks again.

twinbee77, 2021-02-07:
i´m playing a little bit with the esp 2866 version and the esp32 version.
on both boards the mfrc 522 plugin runs fine with the options at this moment.

[50 more replies]

closed 2021-04-15 by TD-er