letscontrolit/ESPEasy #203: Help to compile the new version

ayasystems, 2017-04-06:
Hi

I'm having a problems to compile the new version of EspEasy....

What am i doing wrong? 

Here the error log

https://pastebin.com/S4qHtT07

`ommand:478: error: variable or field 'printDirectory' declared void

 void printDirectory(File dir, int numTabs) {

                     ^

Command:478: error: 'File' was not declared in this scope

Command:478: error: expected primary-expression before 'int'

 void printDirectory(File dir, int numTabs) {

                               ^

Networking:394: error: variable or field 'SSDP_schema' declared void

 void SSDP_schema(WiFiClient client) {

                  ^

Networking:394: error: 'WiFiClient' was not declared in this scope

_N001_Email:112: error: 'WiFiClient' was not declared in this scope

 boolean NPlugin_001_MTA(WiFiClient client, String aStr, String aWaitForPattern)

                         ^

_N001_Email:112: error: expected primary-expression before 'aStr'

 boolean NPlugin_001_MTA(WiFiClient client, String aStr, String aWaitForPattern)

                                                   ^

_N001_Email:112: error: expected primary-expression before 'aWaitForPattern'

 boolean NPlugin_001_MTA(WiFiClient client, String aStr, String aWaitForPattern)

                                                                ^

_N001_Email:112: error: expression list treated as compound expression in initializer [-fpermissive]

 boolean NPlugin_001_MTA(WiFiClient client, String aStr, String aWaitForPattern)

                                                                               ^

C:\Users\Angel\Dropbox\Arduino\ESPeasy\Command.ino: In function 'void ExecuteCommand(byte, const char*)':

Command:49: error: 'RTC' was not declared in this scope

     RTC.factoryResetCounter = 0;

     ^

Command:66: error: 'Settings' was not declared in this scope

       if (Settings.NotificationEnabled[Par1 - 1] && Settings.Notification[Par1 - 1] != 0)

           ^

[... 329 lines cut ...]

                 from c:\users\angel\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\functional:53,

                 from C:\Users\Angel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:27,

                 from C:\Users\Angel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.h:28,

                 from C:\Users\Angel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:34,

                 from C:\Users\Angel\Dropbox\Arduino\ESPeasy\ESPEasy.ino:268:

c:\users\angel\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\exception:35:37: error: expected declaration before end of line

 #pragma GCC visibility push(default)

                                     ^

Se encontraron múltiples librerías para "Servo.h"
Usado: C:\Users\Angel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Servo
 No usado: C:\Program Files (x86)\Arduino\libraries\Servo
Se encontraron múltiples librerías para "SD.h"
Usado: C:\Users\Angel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\SD
 No usado: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
variable or field 'printDirectory' declared void`

krikk, 2017-04-07:
did you follow the guides?
 [https://www.letscontrolit.com/wiki/index.php/ESPEasy#Source_code_development] ?

please try it with the guide...

ayasystems, 2017-04-09:
I'm not ussing platformio. I want compile with Arduino as with old version. Is it possible?

There are any guide to compile with Arduino? 

Regards and thanks!

psy0rz, 2017-04-09:
Please ask your question on the forum.

The issue tracker is for bugs only.

To compile with Arduino IDE follow: https://www.letscontrolit.com/wiki/index.php/Tutorial_Arduino_Firmware_Upload

closed 2017-04-09 by psy0rz