letscontrolit/ESPEasy #205: Feature request - Energy meter SDM120, 220, 630

agsochi, 2017-04-07:
Cheap [Eastron SDM120](https://www.aliexpress.com/item/SDM120C-Single-Phase-Two-Wire-Din-Rail-Energy-Meter-with-RS485-Modbus-and-Pulse-Output-CE/32349129725.html?ws_ab_test=searchweb0_0,searchweb201602_2_10152_10151_10065_10150_10068_10136_10137_10138_10060_10062_10141_10056_10055_10054_10059_10099_10103_10102_10096_10148_120_10147_10052_10053_10050_10107_10142_10051_10143_10084_10083_10119_10080_10082_10081_10110_10111_10112_10113_10114_10037_10078_10079_10077_10073_10070_10123_10120_10124,searchweb201603_6,afswitch_1_afChannel,ppcSwitch_2&btsid=4bc94d05-9780-4aef-97cd-15d5c7ae2ac3&algo_expid=ca5d930f-e6ed-47e5-b7e5-9db243f8da86-0&algo_pvid=ca5d930f-e6ed-47e5-b7e5-9db243f8da86), but good quality.
Can control [13 parameters]( https://drive.google.com/file/d/0Bw1XOPtMPlsNZHJlU2sxZnJ6VWM/view).
[Template library](https://github.com/reaper7/SDM_Energy_Meter) for reading SDM120 / SDM220 / SDM630 Modbus Energy meters:
1$ rs232<->rs485 is needed.

beicnet, 2017-04-08:
@agsochi that is good one, but this is cheaper with Pulse Counter:
[https://www.aliexpress.com/item/5-32-230V-50HZ-Single-phase-Din-rail-Watt-hour-energy-meter-with-LCD-digital-disply/695870527.html](url)

agsochi, 2017-04-08:
This is completely different products like computer and calculator, SDM120 can measure:
```
Voltage -Volts
Current -Amps
Active power -Watts
Apparent power -VA
Reactive power -VAr
Power factor
Frequency -Hz
Import active energy -kWh
Export active energy kWh
Import reactive energy -kvarh
Export reactive energy -kvarh
Total active energy -kWh
Total reactive energy -Kvarh
```
And it stores `Total` values in memory.

beicnet, 2017-04-08:
@agsochi Oh, that is nice then! ;) Thumbs for that!

krikk, 2017-04-08:
i prefer the non-invasive measurement like done with https://guide.openenergymonitor.org/setup/emontx/ more, it is more secure for non electrician like me :)

...so if you want this in espeasy, gear up you coding skills, and submit a pull request...

see here for a start: https://www.letscontrolit.com/wiki/index.php/ESPEasy#Source_code_development

agsochi, 2017-04-23:
Working sketch is [here](https://community.openhab.org/t/how-to-get-a-cheap-smart-meter-with-real-time-power-consumption/7205/11).

sjfaustino, 2017-05-12:
I'm hoping to do a ModBus controller later on so that any device that uses ModBus can more easily be developed.

I'm a newbie with ESPEasy so it's slow going but I currently have a plugin that I'm developing to communicate with ModBus for the Eastron SDM series of Energy meters and despite being in development, it works fine. Is there still an interest?

krikk, Eastron have Energy meters that can use slipt core current transformers.

beicnet, 2017-05-12:
@sjfaustino Great, can you push into the Playground with other plugins?

krikk, 2017-05-12:
@sjfaustino like @beicnet said, please do a pull request if it works and is tested...

beicnet, 2017-05-12:
@krikk I can't do it, because it's not my work! @sjfaustino need to do that!

sjfaustino, 2017-05-12:
There's a pull request on the playground right now.
The plugin works as-is with no issues but it must be developed much further.

My thought is to work with the entire series of SDM with sanity checks (only allow baud rates and query values supported by each meter, do autodetect to check on which pins it's connected (a type of ModBus scan), etc...   

Still a newbie to Arduino C and ESPEasy so if anyone wants to contribute, please do.

sjfaustino, 2017-05-12:
Also, I believe it's more useful to keep it returning a single value since you can either do a dummy device to compile several values or just add the same device several times with the same name.

This allows us to for example only check the total power every hour but check the current every minute.

yuret, 2017-12-15:
Thank you to all who made this possible:)

[image]
[image]

psy0rz, 2017-12-18:
if more people need it and the plugin is clean enough, we might add it to ESPEasy by default.

Please add a thumbs up to this comment if you want that.

TD-er, 2017-12-18:
Hmm, I have a Eastron sensor to monitor my PV, so this is a good one to have.
I was also working on support for other Senseair sensors, which also need Modbus, so I'm all in for separate Modbus support to use with several plugins.

psy0rz, 2017-12-18:
sounds good :)

Arek84, 2018-01-12:
Very good and useful plug.
Does it work with SDM630?

sjfaustino, 2018-01-13:
Yes, ALL the SDM630 values are available. Bit of work since I had to consult 3 datasheets to get them all.
Still plan to improve it a lot but RL got in the way. It's open to anyone who want to have a go at it :)

LeeNX, 2018-01-13:
A little more the $1, but I like the 3.3V support and re-setting fuses - https://www.ebay.com/itm/Stable-RS485-to-TTL-Module-UART-Serial-Port-to-RS485-Converter-Function-Module/361375498598  
  
Ordered and waiting to play, thanks!!

mstjerna, 2018-03-19:
What about this one ? :-)
Schneider iEM3150
https://www.schneider-electric.com/en/product/A9MEM3150/iem3150-energy-meter---63-a---modbus/

sjfaustino, 2018-03-19:
Schneider is a good brand but we are talking about 60€ for a Eastron 630C (been working on a 60KW machine for over 3 years with no issues so far) versus over 200€ for the Schneider one.

If you are going to use it on a production environment where your job is on the line if it fails, a good rule of thumb is _DO NOT use Chinese stuff_, if that's not the case however... :)

mstjerna, 2018-03-19:
How hard would it be to implement with the work already done?
I got a couple of these from a friend, would be nice to fit and use them.

TD-er, 2018-03-19:
I guess we have to compare (programmer) datasheets then to get an idea of the diffs between them.
But I guess it is basically the same. Address the Modbus device, request a register read and interpret the data.
If they list the same output parameters, the plugin could even be shared.

LeeNX, 2018-03-20:
There are some GitHub projects that have the data sheets for the Eastron ModBus meter readings - https://github.com/peninquen/Modbus-Energy-Monitor-Arduino/tree/master/Collection_of_manuals  
  
Was looking at building an ESP/485 Device just for my meters - https://github.com/reaper7/SDM_Energy_Meter  
  
But I would prefer all my ESP device to be running ESPEasy ... I'm just too lazy that way! ;-)

beicnet, 2018-03-20:
> DO NOT use Chinese stuff

Everything is made in China, and no matter what Brand we are talking about... :wink:

Egyras, 2018-04-02:
@yuret  How you managed to include this plugin ?  please share .bin file to upload to NodeMCU 1.0
I have tried to compile with Ardruino IDE.... with no luck

rvmil, 2018-06-05:
I am also trying to get the playground working but with no luck.... Just need the easy esp with this nice plugin. does someone have a compiled version?
Or does someone now where to get a instruction manual to get the playground working.
Maybe some one already seen the error:
Command:42: error: 'Command' does not name a type
 Command commandStringToEnum(const char * cmd) {

Thanks in advance!

erikzw7, 2018-08-22:
For a couple of days now I've been trying to compile this espeasy version for a nodemcu V3 board with the SDM150 plugin but without luck. 
I have taken the following steps:
1. Installed platformio and atom
2. Downloaded the ESPEasy zip ( ESPEasy_mega-20180818)
3. Extracted it all to a directory and compiled it with Platformio. This worked
4. Downloaded ESPEasyPluginPlayground-master.zip and extracted the _P150_SDM120C.ino and copied it to the directory
5. Downloaded SDM_Energy_Meter-master.zip from reaper and extracted the files into the directory
6. When building the file I get the following error:
_P150_SDM120C.ino:86:101: error: no matching function for call to 'addFormSelector(String&, const __FlashStringHelper*, const __FlashStringH
elper*, int, String [10], NULL, byte&)'

Anybody an idea on how to fix this?

TD-er, 2018-08-22:
In the line of that ino file, with `addFormSelector`, remove the first argument (the string)

erikzw7, 2018-08-23:
Removed the string and that seemed to help on that particular problem. However, compiling is still failing, see results below

Compiling .pioenvs\dev_ESP8266_4096\FrameworkArduino\IPAddress.cpp.o
C:/Users/Downloads/ESPEasy_mega-20180818/Source/src/_P150_SDM120C.ino:21:1: error: 'SDM' is not a template
SDM<2400, D6, D7> Plugin_150_SDM;
^
C:/Users/Downloads/ESPEasy_mega-20180818/Source/src/_P150_SDM120C.ino:21:19: error: no matching function for call to 'SDM::SDM()'
SDM<2400, D6, D7> Plugin_150_SDM;
^
C:/Users/Downloads/ESPEasy_mega-20180818/Source/src/_P150_SDM120C.ino:21:19: note: candidates are:
In file included from C:/Users/Downloads/ESPEasy_mega-20180818/Source/src/_P150_SDM120C.ino:20:0:
src/SDM.h:153:5: note: SDM::SDM(SoftwareSerial&, long int, int)
SDM(SoftwareSerial& serial, long baud = SDM_UART_BAUD, int dere_pin = DERE_PIN);
^
src/SDM.h:153:5: note:   candidate expects 3 arguments, 0 provided
src/SDM.h:148:7: note: constexpr SDM::SDM(const SDM&)
class SDM {
^
src/SDM.h:148:7: note:   candidate expects 1 argument, 0 provided
Compiling .pioenvs\dev_ESP8266_4096\FrameworkArduino\MD5Builder.cpp.o
Compiling .pioenvs\dev_ESP8266_4096\FrameworkArduino\Print.cpp.o
*** [.pioenvs\dev_ESP8266_4096\src\ESPEasy.ino.cpp.o] Error 1

LeeNX, 2018-08-23:
I had the above problem, fix that by moving reaper's lib SDM_Energy_Meter into my lib folder and rename to SDM, if I remember correctly.

erikzw7, 2018-08-23:
Tried this, but still without luck: 

Compiling .pioenvs\dev_ESP8266_4096\FrameworkArduino\Print.cpp.o
C:/Users/Annemarie/Downloads/ESPEasy_mega-20180818/Source/src/_P150_SDM120C.ino:21:1:
error: 'SDM' is not a template
SDM<2400, D6, D7> Plugin_150_SDM;
^
C:/Users/Annemarie/Downloads/ESPEasy_mega-20180818/Source/src/_P150_SDM120C.ino:21:19:
error: no matching function for call to 'SDM::SDM()'
SDM<2400, D6, D7> Plugin_150_SDM;
^
C:/Users/Annemarie/Downloads/ESPEasy_mega-20180818/Source/src/_P150_SDM120C.ino:21:19:
note: candidates are:
In file included from
C:/Users/Annemarie/Downloads/ESPEasy_mega-20180818/Source/src/_P150_SDM120C.ino:20:0:
lib\SDM/SDM.h:153:5: note: SDM::SDM(SoftwareSerial&, long int, int)
SDM(SoftwareSerial& serial, long baud = SDM_UART_BAUD, int dere_pin =
DERE_PIN);
^
lib\SDM/SDM.h:153:5: note: candidate expects 3 arguments, 0 provided
lib\SDM/SDM.h:148:7: note: constexpr SDM::SDM(const SDM&)
class SDM {
^
lib\SDM/SDM.h:148:7: note: candidate expects 1 argument, 0 provided
Compiling .pioenvs\dev_ESP8266_4096\FrameworkArduino\Schedule.cpp.o
Compiling
.pioenvs\dev_ESP8266_4096\FrameworkArduino\ScheduledFunctions.cpp.o
Compiling .pioenvs\dev_ESP8266_4096\FrameworkArduino\Stream.cpp.o
Compiling .pioenvs\dev_ESP8266_4096\FrameworkArduino\StreamString.cpp.o
*** [.pioenvs\dev_ESP8266_4096\src\ESPEasy.ino.cpp.o] Error 1

---

 

Links:
------
[1]
https://github.com/letscontrolit/ESPEasy/issues/205#issuecomment-415388632
[2]
https://github.com/

erikzw7, 2018-08-23:
moved the SDM.h file to the lib directory in a separate SDM directory, but this had no effect.

LeeNX, 2018-08-23:
If there more eyes on this, I will have another bash at this and see how I fair, thou I am no developer, so I don't expect much more, other than alot more learning ... :-)

LeeNX, 2018-08-23:
@erikzw7 - Try moving the SDM lib into the Arduino lib folder, thou I am not sure how platform.io deals with external libs?

erikzw7, 2018-08-24:
Made a new directory SDM under libs and the error about not finding the SDM.h is gone, however the following error persists:

============

C:/Users/Annemarie/Downloads/ESPEasy_mega-20180818/Source/src/_P150_SDM120C.ino:21:1: error: 'SDM' is not a template
SDM<2400, D6, D7> Plugin_150_SDM;
^
C:/Users/Annemarie/Downloads/ESPEasy_mega-20180818/Source/src/_P150_SDM120C.ino:21:19: error: no matching function for call to 'SDM::SDM()'
SDM<2400, D6, D7> Plugin_150_SDM;

erikzw7, 2018-08-24:
apparently I'm not the only one with this issue:
http://www.letscontrolit.com/forum/viewtopic.php?f=4&t=5665&start=10

TD-er, 2018-08-24:
I was yesterday already working on it.
The old version of that library used serial port definition as a template.
The new one needs a SoftwareSerial object to interact with.
This conflicts with the ESPEasySoftwareSerial object.

So I will try to solve this issue.

TD-er, 2018-08-24:
Could you please test this pull request: https://github.com/letscontrolit/ESPEasy/pull/1679  ?

erikzw7, 2018-08-26:
> Could you please test this pull request: #1679 [1] ?

Compiling works out nicely without error. However when I upload the new
firmware on the device the wifi does not come on, so not much I can test
further. 

---

 

Links:
------
[1] https://github.com/letscontrolit/ESPEasy/pull/1679
[2]
https://github.com/letscontrolit/ESPEasy/issues/205#issuecomment-415885470
[3]
https://github.com/

erikzw7, 2018-08-27:
> Have you tried the nightly build? (Test build)

Silly me, it worked nicely but it connected immediately to my
accespoint. I thought that after every flash you had to input all the
settings and hence the wifi code would be lost. That is not the case.
Anyway, the 26-08 nightly build compiled and loaded correctly into the
device.  

Next step is to get the hardware connections working.

---

 

Links:
------
[1]
https://github.com/letscontrolit/ESPEasy/issues/205#issuecomment-416056551
[2]
https://github.com/

erikzw7, 2018-08-28:
Managed to upload the firmware. Now connected the Nodemcu to the 485 via D3 and D4 and used this in the ESPEASY webinterface. However I only get NaN for the values of the device.

Findus76, 2018-09-24:
Hello
Has an SDM630. An RS-232 <-> RS-485 is home.
Hope it's just getting download the test version and try it :)
What pin is it for communication and bus turnaround?

Findus76, 2018-09-24:
Looking at the code (P078_Eastron.ino, ESPEasy_mega-20180924) that the Rx is mapped to pin6 and Tx to pin7.
But I find no info about bus turnaround. An RS-485 bus is half duplex. Therefore, a signal from the controller to RS-232 <-> RS-485 is required when switching from transmission to reciving.

seamoor, 2019-07-19:
Hi,
I'm sorry but from what i see this plugin supports only 1 phase power meters like SDM120, 220 etc
I personally have SDM630, this is 3 phase power meter. Connected up and running.
On ESP view: 

[image]
 
We can choose only generic options like Voltage, Current etc but according to SDM.h by Reaper7 this power meter has 3 Voltage registers, 3 Current registers etc:
[image]

How can i select Phase nr 1 Voltage or Phase nr 2 Voltage? 
And now when i select Voltage, output is divided by 3:

[image]

This is wrong result.

TD-er, 2019-07-20:
I guess that's an easy addition to the plugin, just reading another register.
I will look into a way to only list the registers available to the connected node.
It will be not this week though.

seamoor, 2019-07-20:
Thanks,
If you need to test this on real hardware just tell me when, i can flash it and check :)

pite81, 2019-09-14:
> Thank you to all who made this possible:)
> 
> [image]
> 
> [image]

I see that you have several devices in your device list, but what are those? am I right to assume that, you use a plugin?
If so, how do I install it on an esp easy firmware? Thanks.

closed 2019-10-10 by TD-er

seamoor, 2019-10-11:
Hi,

I see that this ticket is now closed.

Has reading from another registers been added? 

> 
> I guess that's an easy addition to the plugin, just reading another register.
> I will look into a way to only list the registers available to the connected node.
> It will be not this week though.

Thanks for your time :)

TD-er, 2019-10-11:
> I see that this ticket is now closed.
> 
> Has reading from another registers been added?

@seamoor I guess not.
Can you add a new issue with just that request?
We (@Grovkillen and me) are closing old issues since lots of them are no longer relevant, duplicates, etc.
Also a lot of issues have grown into threads like "original issue was solved, but the discussion lead to new requests" and those are really hard to maintain.

seamoor, 2019-10-11:
@TD-er sure thing :)
I will create a new issue.

matzeeg3, 2022-06-13:
how can i use the sdm120? what i have to do?
kind regards

tonhuisman, 2022-06-13:
> what i have to do?

- Install an ESPEasy build with `Energy` in the name on your ESP (either ESP8266 or ESP32)
- Connect the SDM120 to that ESP
- Configure a task on the Devices page and enable that
- Test