letscontrolit/ESPEasy #206: Battery drain during sleep mode

rolfz, 2017-04-08:
I am using a couple of EspEasy sensors to measure temperature in my flat.
I noticed that when the ESP module is unable to connect to the router, means no connection in the connect loop is possible, then the module continues to connect until the battery is empty, means within a couple of hours.
It would be great to add a function that tells ESP to go into normal sleep mode (normal delay) after a certain number of attempts which are already included inside the code. In case we measure temperature every 10 minutes, the ESP module would retry to connect x times every 10 minutes and not constantly. This would also mean the battery life would be the same if it connects or not.

psy0rz, 2017-04-08:
that is indeed very bad. this is with 2.0.0-dev7?

rolfz, 2017-04-08:
I do not think there was any change between the last release and 2.0.0-dev7 regarding the "connect" routine. But I will give it a try to be sure with one of my sensors.

rolfz, 2017-04-12:
I confirm, no change with the latest version.

psy0rz, 2017-04-12:
Yep all i needed to know that it was 2.0.0-dev7 or newer. I'll look into it and fix it. (going to use deep sleep on battery's myself as well)

rolfz, 2017-05-23:
Any news on the  battery drain in sleep mode when router is not reachable ?

psy0rz, 2017-05-23:
still didnt get to it:(

psy0rz, 2017-06-01:
working on the wifi stuff right now (making it more efficient and fixing issues)

psy0rz, 2017-06-02:
there where various issues with it. it also was trying to connect to SSID2, even though it wasnt defined. this resulted in 6 connection atteps, which takes ages.

now it will only do one connection attempts on the main SSID, and another one on SSID2 if its defined.

also serial logging is much more clear now.

try it and let me know if you have any issues. :)

rolfz, 2017-06-02:
Thank you ! What version do I need to download to test it ? Would ESPEasy_v2.0.0-dev9_dev_4096.bin make it ?

psy0rz, 2017-06-02:
As far as i could see it would try to connect 6 times, almost taking a minute in total. 

New behaivior: Try to connect 1 time (and 1 time to ssid2 if its defined), do stuff (even if connection fails), sleep again.

I didnt test with any controllers and timeouts for that, so i will test that as well now.

rolfz, 2017-06-02:
Thank you!!

I flashed 3 sensors on lipo with the new version. Need to add some 
battery log on it to check how battery behaves. As of your explanation, 
it should be ok now !!

Many thanks

Rolf

psy0rz, 2017-06-02:
I tested with one sysinfo device to http domoticz, while using a non-existing SSID:
```
INIT : Booting version: 
FS   : Mounting...
FS   : Mount successful, used 72288 bytes of 957314
INIT : Free RAM:25592
INIT : I2C
INIT : SPI not enabled
WIFI : AP Mode disabled
IP   : Static IP :192.168.13.92
WIFI : Connecting blah attempt #1
WIFI : Could not connect to AP!
WIFI : AP Mode enabled, reachable on 192.168.4.1
INIT : Rebooted from deepsleep #37
WD   : Uptime 0 ConnectFailures 0 FreeMem 20320
SYS  : 0.00
HTTP : connection failed
SLEEP: Entering deep sleep...
EVENT: System#Sleep
```

This took almost 20 seconds. So if your sleep delay is 60 seconds, it will have a dutycycle of 25%. 

Now here's what we could do: If it cant connect, just go back to deepsleep immeadeatly without doing anything. I'll think i add an option for this. We should also make controllers smart so that they dont try to connect when there's no connection.

rolfz, 2017-06-02:
ok, I usually use sleep time of 10 minutes (600secs). An option "goto deep sleep if connection fails" would be a good solution!

psy0rz, 2017-06-02:
Ok added a new option for this. will document it on wiki.

closed 2017-06-02 by psy0rz

rolfz, 2017-06-02:
Great, had a question on programming, what environnement are you using 
to develop/fix the code of ESPEASY ?

psy0rz, 2017-06-02:
i'm using Ubuntu 16.04.2 LTS, with platformio.