letscontrolit/ESPEasy #222: Module _P019_PCF8574.ino: error in mixing input pins and output pins

svmac, 2017-04-11:
PCF8574 documentation states "The I/Os should be HIGH before being used as inputs." (https://www.nxp.com/documents/data_sheet/PCF8574.pdf section 7.3).
In function Plugin_019_Write to write an output pin first it reads all pins, then set the pin and write all pins.
If coincide an input pin reads LOW, then writes that pin as LOW and that pin leaves to be input pin. That pin always reads LOW until PCF8574 restart (or until explici writes HIGH to that pin; now I make that on rules after write an output pin).
I think the solution is to keep a mask in the module and set to HIGH all pins defined as inputs. Then in function Plugin_019_Write apply the mask when read all pins, so all input pins are set to HIGH.

psy0rz, 2017-04-12:
Can you try and fix it?

psy0rz, 2018-02-26:
should be fixed, please reopen if this problem still exists.

closed 2018-02-26 by psy0rz