letscontrolit/ESPEasy #225: problem compile withe arduino ide 1.8.2

mihadongle, 2017-04-13:
Arduino: 1.8.2 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

Command:478: error: variable or field 'printDirectory' declared void

 void printDirectory(File dir, int numTabs) {

                     ^

Command:478: error: 'File' was not declared in this scope

Command:478: error: expected primary-expression before 'int'

 void printDirectory(File dir, int numTabs) {

                               ^

Networking:394: error: variable or field 'SSDP_schema' declared void

 void SSDP_schema(WiFiClient client) {

                  ^

Networking:394: error: 'WiFiClient' was not declared in this scope

_N001_Email:110: error: 'WiFiClient' was not declared in this scope

 boolean NPlugin_001_MTA(WiFiClient client, String aStr, String aWaitForPattern)

                         ^

_N001_Email:110: error: expected primary-expression before 'aStr'

 boolean NPlugin_001_MTA(WiFiClient client, String aStr, String aWaitForPattern)

                                                   ^

_N001_Email:110: error: expected primary-expression before 'aWaitForPattern'

 boolean NPlugin_001_MTA(WiFiClient client, String aStr, String aWaitForPattern)

                                                                ^

_N001_Email:110: error: expression list treated as compound expression in initializer [-fpermissive]

 boolean NPlugin_001_MTA(WiFiClient client, String aStr, String aWaitForPattern)

                                                                               ^

C:\Users\Acer\Desktop\Neuer Ordner\ESPEasy\espeasy\Command.ino: In function 'void ExecuteCommand(byte, const char*)':

Command:49: error: 'RTC' was not declared in this scope

     RTC.factoryResetCounter = 0;

     ^

Command:66: error: 'Settings' was not declared in this scope

       if (Settings.NotificationEnabled[Par1 - 1] && Settings.Notification[Par1 - 1] != 0)

           ^

Command:69: error: aggregate 'EventStruct TempEvent' has incomplete type and cannot be defined

         struct EventStruct TempEvent;

                            ^

Command:71: error: 'NPlugin_id' was not declared in this scope

[... 325 lines cut ...]

                 from C:\Users\Acer\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.h:28,

                 from C:\Users\Acer\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:34,

                 from C:\Users\Acer\Desktop\Neuer Ordner\ESPEasy\espeasy\ESPEasy.ino:268:

c:\users\acer\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\exception:35:37: error: expected declaration before end of line

 #pragma GCC visibility push(default)

                                     ^

Mehrere Bibliotheken wurden für "SD.h" gefunden
 Benutzt: C:\Users\Acer\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\SD
 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\SD
Mehrere Bibliotheken wurden für "Servo.h" gefunden
 Benutzt: C:\Users\Acer\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Servo
 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\Servo
exit status 1
variable or field 'printDirectory' declared void

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

mihadongle, 2017-04-13:
now i have tested with the 1.6.12 also the same

beicnet, 2017-04-13:
I don't have any problems with 1.8.1

mihadongle, 2017-04-13:
i have no idea why!!!!!!!!

psy0rz, 2017-04-13:
please read the guide lines for this issue tracker and go to the forum for support.

closed 2017-04-13 by psy0rz

krikk, 2017-04-13:
and just to note, compile DOES work with arduino 1.8.2!

mihadongle, 2017-04-13:
with my pc it doesn´t

jjansen85, 2018-01-11:
Got this error myself. Just for further documentation; when your folder-name is not exactly **ESPEasy** you will get this error. In this issue the folder name appears to be **espeasy**.

I will include this in the [WIKI-page](https://www.letscontrolit.com/wiki/index.php/Tutorial_Arduino_Firmware_Upload)  for documentation.