letscontrolit/ESPEasy #230: Daily flash rate exceeded.. urgent.

jroux1, 2017-04-18:
### Steps to reproduce
I have multiple Dvices, and complex rules, just too many config changes for 2.0.0.Dev7 to handle in the daily flash rate limit. I waited a day, tried to config the last outstanding device, 'daily flash rate exceeded' still active

### Expected behavior
Save to flash, for more than the usual Device configs.

### Actual behavior
Not saving Device config of the Value Names, and the Device, to flash.

### System configuration
1 relay using Switch, 2 pulse counters, one DHT22, 2 System Info devices, Lum TLS2561 SDA & SLC, DS18b20
Hardware:
Wemos D1 mini

Software or git version:
GIT version:	v2.0.0-dev7

jroux1, 2017-04-18:
Found a work around, a power off/on cycle sorts it out.

psy0rz, 2017-04-18:
yes a powercycle resets it. we will adres this issue later. thanks for making the ticket.

psy0rz, 2017-04-19:
we should atleast add a note to this errror message, saying that the counter can be reset with a power cycle.

psy0rz, 2017-05-31:
Ok i've added a message that tells the user to powercycle.

psy0rz, 2017-05-31:
You can also change MAX_FLASHWRITES_PER_DAY in ESPEasy.ino. Default its 100.

If someone thinks this should be higher for practical use, let me know and reopen the ticket. i'll close this for now.

closed 2017-05-31 by psy0rz