letscontrolit/ESPEasy #232: NDIR sensor in relation to Domoticz controller

rvakkeren, 2017-04-18:
### Steps to reproduce
Using ESP Easy with the NDIR MH-Z19 plugin

### Expected behavior
Domoticz should be updated, but it only updates the sensor with a zero-value.

### Actual behavior
The Domoticz API requires the following syntax for a Air Quality sensor  "/json.htm?type=command&param=udevice&idx=IDX&nvalue=PPM". 

Instead of a nvalue, a svalue is used by ESP Easy. That's why the data isn't processed in Domoticz.

psy0rz, 2017-04-19:
i have this sensor myself and will test (and fix) this in the future.

remyderuysscher, 2017-11-19:
Wow, this bug is pretty severe. Can't believe it's still open.

TD-er, 2017-11-19:
If this is still an issue, I will write a fix for it.
Is it an idea to add the option to select the output format?

TD-er, 2017-11-19:
Just a question... Does the Senseair sensor work like expected?
It looks like both Domoticz controllers output only "&svalue=" statements.
Controller Plugin 001 (Domoticz HTTP) does not even contain the "nvalue" string.
Controller Plugin 002 (Domoticz MQTT) does only output a '0' for "nvalue".

So this issue is bigger than only adding an option to a single plugin. It will be an extension to at least both Domoticz controllers and perhaps even others and IMHO is that something for after the 2.0 release.

Grovkillen, 2017-11-20:
Yes I believe that this is a revamp of the way we need to publish to Domoticz. A thing for 2.1 for sure.

psy0rz, 2017-11-20:
@remyderuysscher i'm indeed using this sensor myself, but i added it as "Custom sensor". Then it works without problems.

Maybe its enough to just update the documentation on the wiki to explain this?

remyderuysscher, 2017-11-21:
I'm still not getting any reading in domoticz with a custom sensor type using the REST API.

Grovkillen, 2017-11-21:
Not with custom publish using rules either?

remyderuysscher, 2017-11-21:
What kinds of rules should I be using?

TD-er, 2017-11-21:
Just add it as a "Custom sensor", via the "Dummy" (Virtual) devices.
Then select Domoticz HTTP as controller and couple that to the sensor device.
Then you can use these sensors in Domoticz.
I've several of them working here.

remyderuysscher, 2017-11-22:
Thanks, works now with a custom sensor. Will try it with rules and type CO2
sensor later...

svde, 2018-05-14:
I use this rule:

on CO2#PPM do
  SendToHTTP domoticz-fqdn,80,/json.htm?type=command&param=udevice&idx=228&nvalue=[CO2#PPM]
endon

sincze, 2019-03-19:
on CO2#PPM do
SendToHTTP domoticz-fqdn,80,/json.htm?type=command&param=udevice&idx=228&nvalue=[CO2#PPM]
endon

i used that rule as well, It works. 
However now I changed to Domoticz that requires authentication.... and it does not work anymore. 

(ESP_Easy_mega-20190315_normal_ESP8266_4M.bin)

Created Controllers (Domoticz HTTP)
I tried with Device - Generic Dummy Device (SENSOR_TYPE_SINGLE).
But it seems it is unable to call a /json.htm?type=command&param=udevice&idx=IDX&nvalue=PPM

Generic Dummy that I use for Temperature  work correctly
/json.htm?type=command&param=udevice&idx=IDX&nvalue=0&svalue=TEMP

TD-er, 2019-03-19:
As far as I know, all data we send to Domoticz in one of both controllers only send "svalue" data, not "nvalue".
So I just use a "custom sensor" in Domoticz and send my CO2 values to that IDX and it sends it as "svalue" since we do not yet support sending "nvalues"

sincze, 2019-03-20:
That would indeed be a nice work around and nvalue feature request :).

sincze, 2019-03-20:
Jups it does the trick,
[image]

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

TD-er, 2023-09-03:
Nope, this still is not (yet) finished.