letscontrolit/ESPEasy #235: save errors to flash are not giving details in webinterface.

psy0rz, 2017-04-19:
SaveToFile()should return an error string which in turn can be shown in the webgui and others.

lucaberta, 2017-04-30:
I too am running into the "save to flash error" and it's not clear what this means. Some more clarity is definitely needed!

psy0rz, 2017-07-17:
proving to be a bit more work than expected, adding error checking everywhere and cleanup flashcounting and webgui-feedback

psy0rz, 2017-07-17:
already discovered an important bug by these new checks: security.dat was 512 bytes  but this is too small for current security struct.

psy0rz, 2017-09-23:
still have to test this, for now: if i made a mistake, major stuff will be broken :)

psy0rz, 2017-09-23:
@mvdbro tomorrow i will add -Wall and -Werror to the compiler flag to fix lots of obscure errors. for example:

`warning: control reaches end of non-void function [-Wreturn-type]`

because warnings where off, i didnt get this error, and i had some strange crashes. for example adding 
`if (false) { return(""); }` crashed, and removing that line didnt crash. after adding a return("") to the end of the function things where fine. that took some hours to debug :P

psy0rz, 2017-11-15:
saw another issue with save errors, will look at that one.

closed 2017-11-15 by psy0rz