letscontrolit/ESPEasy #239: ESP reboots after connecting to Generic HTTP controller

rvakkeren, 2017-04-20:
### Steps to reproduce
When using a Generic HTTP controller instead of Domoticz controller

### Actual behavior
I want to use the Generic HTTP controller instead of the Domoticz controller. But everytime the ESP wants to connect to the raspberry (via the generic http controller), it reboots. Disabling the controller solves this problem, but the data won't be sent. 

### System configuration
Hardware: ESP 12-F, MH-Z19
Software or git version: Mega branch

[image]

psy0rz, 2017-04-20:
can you try again with the latest git-checkout? are you using generic http or generic http advanced?

which version did you test?

rvakkeren, 2017-04-20:
Yesterday, I cloned the version and made a build, using PlatformIO, 

Build: 20000
Version: ESP Easy Mega

[image]

thymjan, 2017-04-20:
Same behavior with FHEM. But it depends on the configured devices. Did you test it without any configured device?

rvakkeren, 2017-04-20:
Yup... I can set thecontroller ánd/or device enabled. But as soon as I enable the controller at the device section, it´ll reboot as soon as it tries to connect.

psy0rz, 2017-04-20:
then we might just broke stuff.. can you try again with the official dev7 release?

rvakkeren, 2017-04-20:
will try that tonight, keep you posted.

thymjan, 2017-04-20:
I can confirm that interfacing with FHEM *is* working with the official dev7 release. With the actual (today) cloned version it does not (also tested with just one enabled system "load" device). ESPEasy crashes after enabling the transfer and loops with booting after receiving the first event.
[bootloop.txt](https://github.com/letscontrolit/ESPEasy/files/945272/bootloop.txt)

rvakkeren, 2017-04-20:
Hmm, for some reason, Atom freezes when I try to compile the dev7
release...

thymjan, 2017-04-20:
You don't have to compile dev7 yourself, just get it compiled here:
https://github.com/letscontrolit/ESPEasy/releases
(you hopefully have a windows 10 machine?)

rvakkeren, 2017-04-20:
That one doesn't have the NDIR CO2 sensor in it...

thymjan, 2017-04-20:
Just give it a try without the special sensor. Upload firmware dev7. Define a device like system uptime. Start your http controller. This should work.
Then go back to the actual version, do the same.
I'll expect the controller will not work with just this simple unique device. That's the thing to find out. If this should work, than try this with your special sensor.

psy0rz, 2017-04-23:
fixed

commit 9e53288c (2017-04-23): fixed #239 due to improper PSTR. cleaned up string handling
closed 2017-04-23 by psy0rz