letscontrolit/ESPEasy #240: _P047_i2c-soil-moisture-sensor.ino higher moisture values

moorjunge, 2017-04-20:
### Steps to reproduce
I had some sensors that give values over 800 on moisture, up to 1100.
So i get an info "SoilMoisture: Bad Reading, resetting Sensor..." and nothing is send to the controller.

After change the line 174 at _P047_i2c-soil-moisture-sensor.ino to 
     if (temperature>100 || temperature < -40 || moisture > 1100 || moisture < 1 || light > 65535 || light < 0) {

everything seems to be fine

### Expected behavior
Tell us what should happen?

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:

Software or git version:

#define ESP_PROJECT_PID           2016110801L
#define VERSION                             2
#define BUILD                           20000 // git version 2.0.0
#define BUILD_NOTES                 " - Mega"

//build all plugins that are in test stadium
#define PLUGIN_BUILD_TESTING

krikk, 2017-04-20:
what reading do you get if you put the sensor in a glass of water?? and what do you get in dry air?  ...1100 seems a bit high to me, never get more than 500, so the limit of 800 should be sufficient...

moorjunge, 2017-04-21:
Here is a Screenshot with two sensors in moist soil, but i will also do
some more tests at the Weekend.

krikk, 2017-04-27:
please provide more info, otherwise this issue can be closed...

moorjunge, 2017-05-01:
Hello,

it seems, that all my sensors are broken.
I just ordered two new one.

So, i will close the issue.

Greetings Thiemo

closed 2017-05-01 by moorjunge