letscontrolit/ESPEasy #241: Settings - Save. Filename more descriptive

jroux1, 2017-04-21:
### Steps to reproduce
How can we trigger this problem?
- Save the settings on the Tools menu

### Expected behavior
Tell us what should happen?
- The file name is 'config.dat'

### Actual behavior
Tell us what happens instead?
- As before, lets make the file name the name of the ESP a per config, and a date, that way its easier to store many ESP's configs. 
- example: 'ESP62_2017.04.21.dat'

### System configuration
n/a
Hardware:

Software or git version:
v2.0.0-dev7

sjfaustino, 2017-05-06:
On file Webserver replace the function handle_download with:

void handle_download()
{
  if (!isLoggedIn()) return;

  fs::File dataFile = SPIFFS.open("config.dat", "r");
  if (!dataFile)
    return;

  String str = F("attachment; filename=");
  str += Settings.Name;
  str += "_";
  if (Settings.UseNTP)
  {
    str += year();
    if (month() < 10)
      str += "0";
    str += month();
    if (day() < 10)
      str += F("0");
    str += day();
  }
  str += (".dat");

  WebServer.sendHeader("Content-Disposition", str);
  WebServer.streamFile(dataFile, "application/octet-stream");
}

CCOSTAN, 2017-05-07:
I think part of the issue here is a restore has to be called config.dat. So if the save is changed, the restore needs to be fixed as well to accept new naming conventions.

psy0rz, 2017-05-10:
well this issue is fixed. indeed you need to rename the file to config.dat again when uploading, but a warning is shown for this in the webinterface.

perhaps we can make a new change that fixes it so that it accepts all filenames. (or at least looks for the string 'config' or something)

closed 2017-05-10 by psy0rz