letscontrolit/ESPEasy #244: Unable to find current revision in submodule path 'lib/IRremoteESP8266'

nl0pvm, 2017-04-23:
### Steps to reproduce
Execute the in the readme.md written command: 
`Unable to find current revision in submodule path 'lib/IRremoteESP8266'`

### Expected behavior
The 'lib/IRremoteESP8266' should have been initalized

### Actual behavior

```
\ESPEasy>git submodule update --checkout --init
fatal: Needed a single revision
Unable to find current revision in submodule path 'lib/IRremoteESP8266'
```

### System configuration
Hardware: esp2866
Software or git version: Windows 10 ;-)
Latest commit version: `d1180e4e478e4582fd2f18b1ca8d2f0aa8350e72`

nl0pvm, 2017-04-23:
Without this module I ran into build errors, I have to continue ;)

1. I removed the 'lib/IRremoteESP8266'
Then I ran the command below:
```
\ESPEasy>git submodule update --checkout --init
error: no such remote ref a7ad7b08b67763eb1efe5e97b199a88e471fcd9b
Fetched in submodule path 'lib/IRremoteESP8266', but it did not contain a7ad7b08b67763eb1efe5e97b199a88e471fcd9b. Direct fetching of that commit failed.
```
Not fixed yet ;)

nl0pvm, 2017-04-23:
1. I removed the 'lib/IRremoteESP8266'

2. Then I ran the command below: 
```
\ESPEasy\lib>git clone https://github.com/markszabo/IRremoteESP8266.git
Cloning into 'IRremoteESP8266'...
remote: Counting objects: 684, done.
remote: Compressing objects: 100% (45/45), done.
remote: Total 684 (delta 28), reused 0 (delta 0), pack-reused 636
Receiving objects: 100% (684/684), 311.64 KiB | 0 bytes/s, done.
Resolving deltas: 100% (416/416), done.
Checking connectivity... done.
```

Fixed

psy0rz, 2017-04-23:
will prevent this issue in the future by removing submodules and adding files statically in lib/

psy0rz, 2017-04-23:
fixed in latest commit. 

to get rid of submodules, do this one time:

* delete the lib-directory
* then do:
`git pull`
`git checkout lib`

closed 2017-04-23 by psy0rz

nl0pvm, 2017-04-23:
Myth confirmed; Bug fixed; Thanks