letscontrolit/ESPEasy #252: MQTT Import - no worky, random values and ADC (TOUT)

JonSilver, 2017-05-01:
### Steps to reproduce
I freshly flashed a Wemos D1 Mini with Release v2.0.0-dev8 (Normal). I configured it with an OpenHab MQTT controller. I added three devices - one framed OLED and two MQTT Import devices. 

### Expected behavior
The MQTT Import devices should subscribe to their configured topics and display published data. The devices screen should show no associated GPIO pin.

### Actual behavior
The MQTT Import devices do not appear to receive data from their subscription topics, but instead display spurious values. The devices screen displays ADC (TOUT) for GPIO pin instead of blank.

### System configuration

Hardware:
Wemos D1 Mini
OLED display (128x64 i2c)

Software or git version:
Release v2.0.0-dev8 (Normal)

psy0rz, 2017-05-04:
mqtt code needs some extra love and attention. will probably get to that after we release 2.0.0, because there are a lot of  fuzzy things hacks with the current mqtt stuff everywhere.

checking all the mqtt code and commenting everything probably helps also.

i know there are some "temporary" hacks in domoticz mqtt for example. we need to make that non-hackish and work the same for other mqtt controllers as well.

also i didnt even know we had a mqtt import device. :)

JonSilver, 2017-05-04:
Thanks for the answer @psy0rz - I thought that might be the case. If I have any spare time at all I'd offer to take a look - alas I'm over-committed as it is. So I really appreciate the time and effort you guys are putting into this thing.

The MQTT Import plugin started life as a Plugin Playground project of namirda, a companion to the Framed OLED plugin. I guess someone in the team thought it'd be useful incorporated into the core of the Mega/v2 branch. It's incredibly useful!

starfish1107, 2017-06-03:
some of the mqtt functionality in Namirdas plugins has been blocked intentionally see [https://www.letscontrolit.com/forum/viewtopic.php?f=18&t=2257&start=150#p15890](url)  maybe your problem is related to this.

mvdbro, 2017-06-11:
Fixed the incorrect ADC (TOUT) display. The plugin had a wrong device type set.
Can't reproduce the reading errors here, but i don't have OpenHAB. I did run a few tests using MQTTfx tool and that seems to work as expected. You could check the logging to see what's going on inside the ESP...

JonSilver, 2017-06-11:
Thanks, I shall go and try it over the next week and report back.

Grovkillen, 2018-04-11:
This issue is happening due to more than one MQTT Import?

TD-er, 2018-04-11:
I've had some report on the forum that recent builds now support multiple MQTT imports.
Not verified myself, but at least he was happy :)
There have been some changes on the MQTT import lately, but I did not expect this to be fixed already.

Grovkillen, 2018-04-11:
Okay, I was doing some backlog reading and thought this might be on the horizon of fixed issues...

Grovkillen, 2018-10-23:
We close this one, please reopen if it's still an issue.

closed 2018-10-23 by Grovkillen