letscontrolit/ESPEasy #261: HTTP Advanced no Hosname IP resolve

zetrix1, 2017-05-03:
### Steps to reproduce
Using hostname configuration in http advanced, making a post to a server, if the server changes his IP. The espEasy can´t connect again. 

Powering off and on don´t resolve nothing.
Only when disable and enable controller it resolve IP hostname

### Expected behavior
When a communication is initiated the esp should resolve the hostname IP.

### Actual behavior
The hostname IP is not resolved on every comunication

### System configuration

Hardware:
wemos D1 mini
EspEasy v2.0.0-dev8

psy0rz, 2017-05-03:
yeah this is actually a "feature". it only resolves the hostname when you save the controller settings.

this is probably done to prevent resolving the hostname every time it has to send data.

i will try to fix something so that it resolves the hostsnames on every powerup or something.

will have to think about it a while since we dont want to do it any resolving after resuming from deepsleep.

Edwin

krikk, 2017-05-03:
@psy0rz if we change this we should do it right, and doing dns lookups only on bootup or on creating the controller is both WRONG, the only right solution would be to resolve the host according to the TTL info in the DNS record, but i don't think we get the ttl info from the resolver, so the only solution which we can implement and which would not make troubles, would be to do a DNS lookup everytime!  why not? ...more cpu required, a bit more network traffic...

psy0rz, 2017-05-03:
you're right..and if someone doesnt want dns lookups (uses less time/power for deep sleep nodes?) they can always just use an hardcoded IP.

I agree the current way is quite hackish.

psy0rz, 2017-05-03:
altough i would like to postpone this change until we've solved the current stability issues. doing a lookup every time is a pretty big change that might give us new issues.

ryanjjennings, 2017-05-04:
I would like to put my vote in for resolving more frequently.  My situation is running in a remote location uploading temperature to weather underground.  They seem to periodically change the server so my system works until they change it, then it's broken til I get espeasy to resolve again.  I would prefer it resolve every use or recognize an error and then resolve.  

I am not sure what other people are doing, but my initial impression was if you didn't want it to resolve every time you did it once manually and then used that ip rather than a hostname.

Hopefully that didn't come across critical.  You guys have done an amazing job on this project.  I just wanted to get my use case out there.  Thanks!

Ryan

psy0rz, 2017-05-04:
Ryan, I think we will indeed resolve DNS every time a request is done.

uzi18, 2017-05-05:
Originally it was checkbox there in controller options to resolve or not.
It was named badly - use hostname, but if you change it to use DNS,  it
will be better.

krikk, 2017-05-26:
@zetrix1 can you please test this, if it works like expected, and give feedback if not, thanks

zetrix1, 2017-06-03:
@krikk Yes It works! Great job it works perfectly. All this code is fabulous, thanks

at4500, 2017-09-13:
I have serious doubts issue being fixed. After router obtains new IP address, remote internet nodeMCU still sending to old address. In Controllers tab, the address is "first time" resolved IP address, not the actual DNS.
In older releases, command initiated reboot helped to resolve this issue. Now, only replacing the dns name by same one helps to re initiate the connection

krikk, 2017-09-15:
this is fixed in http advanced controller, but not in the other controllers!

krikk, 2017-09-15:
@at4500: did you use the http advanced controller???

psy0rz, 2017-09-15:
oh right. was reading on my phone while waiting :)

k-sartor, 2017-09-30:
Is it possible to extend the workaround to each controller (in my case domoticz one)?
Or what is the file to change? TY

psy0rz, 2017-10-01:
@k-sartor there might be an issue for it already, if not: please create one.

closed 2017-10-01 by psy0rz