letscontrolit/ESPEasy #275: Settings - Save doesn't save entire configuration

sjfaustino, 2017-05-07:
### Steps to reproduce
Configure SSID, WPAKey, fixed IP info on optional settings, rules
Go to Tools, and click on the Save button in front of settings
compile a new version so that there is a format of the config
Load the file saved on the first step

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
yes

### Expected behavior
Many more settings should be saved on the config.dat file

### Actual behavior
Only some settings are saved

krikk, 2017-05-08:
which settings are not saved??

because these settings are excluded by default:
  char          WifiSSID[32];
  char          WifiKey[64];
  char          WifiSSID2[32];
  char          WifiKey2[64];
  char          WifiAPKey[64];
  char          ControllerUser[CONTROLLER_MAX][26];
  char          ControllerPassword[CONTROLLER_MAX][64];
  char          Password[26];

in current mega build you can save this when you export the security.dat file....

psy0rz, 2017-05-08:
yes this is a feature

sjfaustino, 2017-05-10:
Rules are not saved and I would assume those are part of the configuration

psy0rz, 2017-05-10:
Ah yes this is a feature as well, but should be documented better. If you browse the flash files via the webgui, you will see .txt files for the 4 rules sections.

So config only saves config settings, but NOT rules and security settings. I'll add a documentation tag to this issue.

Grovkillen, 2017-05-28:
I will make this much more clear in the wiki. We're doing a clean up of the main page which will make the future documentation easier to maintain.

psy0rz, 2017-05-28:
very nice grovkillen! i saw you already made lots of improvements to the wiki. i appreciate it.

could you also move the ugly css examples somewhere else? i dont know if they are relevant anymore since @jkDesignDE  made some improvements and changes as well.

@jkDesignDE you already made some documentation somewhere, right?

Grovkillen, 2017-05-28:
Yeah I hear that, the css example will be moved.

psy0rz, 2017-05-28:
cool thanks

jkDesignDE, 2017-05-28:
I will insert the new CSS example and template example if page is moved.

May be a combined page like "Cusomizing Web Front End"

psy0rz, 2017-05-28:
thanks, sounds good! :)

Grovkillen, 2017-05-29:
https://www.letscontrolit.com/wiki/index.php/ESP_web_interface_customization

Is now live.

Grovkillen, 2017-05-29:
I consider this as closed now:

https://www.letscontrolit.com/wiki/index.php/ESP_Easy_web_interface#Backup_and_restore

closed 2017-05-29 by psy0rz