letscontrolit/ESPEasy #276: Bug on ArduinoEasy

annak1n, 2017-05-08:
Hi i have some problem. 
I download on gidhub source and uploading on Mega 2560+W5100
ArduinoEasy start all good, but have 1 bug, then i try to add new Device in menu Devices i can`t.
I'm trying to select a switch for example, but when I select it, nothing happens.

UPD: 
I found out that not on one page of ArduinoEasy you can not choose anything, it just is not saved.
But if I download the firmware from here: https://www.letscontrolit.com/wiki/inde ... ak_preview
And upload it into Mega 2560 then everything works.
Who can answer what the problem is.

krikk, 2017-05-08:
this is about ESPEasy, the repository for ArduinoEasy is over here: [https://github.com/letscontrolit/ArduinoEasy](url) ...so please post this there...

annak1n, 2017-05-08:
1) Link its broken
2) I need to upload ESPEasy to Arduino Mega+W5100 not in ESP8266

psy0rz, 2017-05-08:
arduino mega is not supported anymore

closed 2017-05-08 by psy0rz