letscontrolit/ESPEasy #277: Feature request: tcp/ip Ping test

Tony-in-Oz, 2017-05-09:
With so many modules around the house and the likelyhood modules can stop publishing/sending data for various reasons including power/battery issues, I'd be keen to see a module/device to monitor the ESPeasy modules.
I have 2 in production and 3 in test, 3 went uncontactable and hence un noticed today. Hence of interest to me. I'm thinking of a Ping test with results response time and missed replies.  Somehow this would then be used by the Domoticz server producing a message, email etc.
A full blown Nagios server is likely an over kill, but an extremely cut down basic, ping test, fail, alert would hopefully advise of module issues.

Tony

psy0rz, 2017-05-09:
can you just do a ICMP ping with some domoticz timer script?

psy0rz, 2017-05-10:
So you need some tool that monitors your ESPEasy. Monitoring can be done by letting the tool do an ICMP check or perhaps an http-request to the ESPEasy webinterface.

I dont think you need a new feature for that in ESPEasy, you just need a monitoring tool.

closed 2017-05-10 by psy0rz