letscontrolit/ESPEasy #284: DHT plugin suggested code revision to fix 'NaN' issue due to incorrect I/O pin control

linker3000, 2017-05-10:
### Steps to reproduce
Attach DHT11/22 sensor to an ESP8266 board running ESPEasy and monitor the MQTT messages generated.

Does the problem persist after powering off and on? (just resetting isnt enough sometimes) - YES

### Expected behavior
100% valid temperature and humidity readings.

### Actual behavior
A high number (about 5%) of messages return values as NaN

### System configuration

Hardware:
Wemos D1 mini (issue is not specific to this platform).

Software or git version:
R1.47RC8 but specific to the DHT plugin module _P005_DHT

Notes:

With reference to: "DHT init sequence wrong" : https://github.com/adafruit/DHT-sensor-library/issues/48

"...after you pulled the data line low for the at least 18 milliseconds to signal the DHT11 a start, you should NOT actively pull it up. As soon as the DHT does see the low-high transition on the data line ( Well "as soon is around 12 usec later) it tries to pull it down. If the data line is still held "high" by the uC at that time, you have more or less a short circuit on the data line with the uC pulling up and the DHT pulling down."

The ESPEasy DHT plugin code seems based on the Adafruit code, so this issue is relevant and suggests that one line in the plugin needs removing or commenting out:

Line 100...

```
        digitalWrite(Plugin_005_DHT_Pin, LOW);              // Pull low
        delay(18);
//        digitalWrite(Plugin_005_DHT_Pin, HIGH);             // Pull high <<< Remove/Comment out
        delayMicroseconds(20); // was 40
```
I have confirmed that recompiling and running this plugin with the above line removed gives me 100% good readings with a DHT11 so I suggest further validation is done and the change incorporated into the live code.

linker3000, 2017-05-10:
Thanks for the quick response. I suppose the 20 microsecond delay right after the 18mS one is a bit superfluous too, but there is no harm leaving it in unless code space is really really tight!

krikk, 2017-05-29:
i will revert this change, because with this "fix" i get way more errors than without it... did you test this, and how? ..i tested with 2 dht22 attached to a wemos d1 r2...

jctual, 2018-01-17:
The suggestion to remove the digitalwrite HIGH is very sound.
However after setting the sensor as INPUT, you also need to add delayMicroseconds(50);
I am currently updating the DHT plugin to include sonoff am2301 and si7021.
It works fine on sonoff am2301 and si7021 after removing the digitalwrite HIGH.
If interested my code is attached. 
Unfortunately I cannot test for DHT11, DHT12 and DHT22.
[_P005_DHT.zip](https://github.com/letscontrolit/ESPEasy/files/1639911/_P005_DHT.zip)

psy0rz, 2018-01-18:
hopefully fixed with #736. also #701 makes sure that NaNs will never be send to controllers, so you will be missing values, but your controller wont show errors or get confused.

please reopen if its still a problem.

closed 2018-01-18 by psy0rz

TD-er, 2018-09-24:
Since this is an old thread, could you add a new one with your findings and refer to this thread?
Old threads are rather hard to find, especially when they are closed.