letscontrolit/ESPEasy #293: json output nan values not double quoted

pmitev, 2017-05-13:
### Steps to reproduce
How can we trigger this problem?
I have a "AM2302" sensor. When the sensor is disconnected, the json prints **nan** without double quotes for the missing values:

```
...
"Sensors":[
{
"TaskName": "AM2302",
"Temperature": nan,
"Humidity": nan
},
...
```

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

It persists.

### Expected behavior
Tell us what should happen?

In this particular case **nan** should be enclosed in double quotes - **"nan"**

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:
NodeMCU v0.9
Software or git version:
v2.0.0-dev8

beicnet, 2017-05-14:
@pmitev True, else it wont be proper (not a valid) JSON output

Grovkillen, 2018-03-01:
Is this fixed?

pmitev, 2018-03-02:
It is still not fixed (I have tried it with some of the latest releases).

TD-er, 2018-03-02:
If it is was still an issue with the builds of the last week, it is still an issue.
I added an isFloat() function, which could be used to wrap a string with quotes. That function also allows strings which represent an int.

pmitev, 2018-03-02:
still no joy
{"System":{
"Name":"Node_1","Unit":1,"Build":20100,"Git Build":"**mega-20180302**","Local time":"2018-03-02 18:50:22","Uptime":3,"Free RAM":14320},
"Sensors":[
{
"TaskName":"DHT22",
"T": **nan**,
"H": **nan**
},

TD-er, 2018-03-02:
I will fix it this evening.

---
Verzonden vanaf laptop

uzi18, 2018-03-02:
Why not to add double quote for every value?

TD-er, 2018-03-02:
Because some expect an integer and integers don't have quotes.

TD-er, 2018-03-02:
See PR #990
The output of /json will be formatted like this:
```json
{"System":{
"Build":20000,
"Git Build":"",
"Local time":"2018-03-02 21:48:07",
"Unit":0,
"Uptime":0,
"Free RAM":16072
},
"Sensors":[
{
"tasknr":1,
"TaskName":"temp",
"Temperature":23.80,
"Humidity":19.95,
"Pressure":1003.63
},
{
"tasknr":2,
"TaskName":"oled"
},
{
"tasknr":3,
"TaskName":"free",
"ram":16280
},
{
"tasknr":4,
"TaskName":"switch",
"Switch":1
},
{
"tasknr":5,
"TaskName":"testnan",
"test":"inf"
},
{
"tasknr":6,
"TaskName":"",
"Output":0
}
]}
```

pmitev, 2018-03-03:
Now it works.

{"System":{
"Build":20100,
"Git Build":"**mega-20180303**",
"Local time":"2018-03-03 07:52:02",
"Unit":1,
"Uptime":1,
"Free RAM":14440
},
"Sensors":[
{
"tasknr":1,
"TaskName":"DHT22",
"Type":"Environment - DHT11/12/22  SONOFF2301/7021",
"T":**"nan"**,
"H":**"nan"**
},

TD-er, 2018-03-03:
Great!

closed 2018-03-03 by TD-er

beicnet, 2018-03-03:
So, what is that sensor type at the end `"Type":"Environment - DHT11/12/22 SONOFF2301/7021",`?

TD-er, 2018-03-03:
@beicnet That's the label of the selected plugin, also shown in the pull down selection when you add a plugin.

beicnet, 2018-03-03:
@TD-er Yes, I know that, I was asking for the last sensor type **7021**, what is that, what manufacturer?
Because, if we talk here about the Si7021 then that is in wrong selection, if not, then it will confuse a lot of users by the mark!

TD-er, 2018-03-03:
No idea, we'll have to look into that (probably you can do that also :) )
If it should be changed, please add an issue for it.

beicnet, 2018-03-03:
Yeah! :laughing: As you know Si7021 is an I2C device, and can't be any near of that plugin (if it's a Si7021)

beicnet, 2018-03-03:
@TD-er what I have found out is that the current pugin name is for **"custom made"** ITEAD Sonoff Sensor Si7021

So, there is the bare Si7021 which is an I2C device and there is the ITEAD provided Si7021 external sensor which is the same I2C device controlled by an **internal microcontroller** to make it a **one wire device**.

So, the current plugin [_P005_DHT.ino](https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P005_DHT.ino) selection/type/name will mislead users for shoure!

Just my 5 cents! :wink: Kappa

pmitev, 2018-03-03:
the one I have hooked is DHT22/AM2302 https://cdn.shopify.com/s/files/1/1978/9859/products/04_46_1024x1024.jpg