letscontrolit/ESPEasy #294: ESPEasy/src/_P036_FrameOLED.ino No such file or directory #include "images.h"

Vladnwx, 2017-05-13:
### Steps to reproduce
How can we trigger this problem?
GIT REPO, Install by https://www.letscontrolit.com/wiki/index.php/Tutorial_Arduino_Firmware_Upload

The compiler can not find the file #include "images.h"

### Expected behavior
firmware Build

### Actual behavior

The compiler can not find the file #include "images.h"

ESPEasy/src/_P036_FrameOLED.ino No such file or directory #include "images.h"

### System configuration

Hardware: GENERIC ESP8266 MODULE

Software or git version: Arduino IDE 1.8.2

krikk, 2017-05-13:
The lib folder has all the libraries you need, did you copy them to the Arduino libraries folder?

Vladnwx, 2017-05-13:
Thank you. I downloaded libraries from arduino ide, which did not exist, copied from the folder. Copied everything from the folder - worked

closed 2017-05-13 by Vladnwx