letscontrolit/ESPEasy #303: OTA failed dev007->dev008 on Sonoff S20 plug (1MB)

hphde, 2017-05-17:
### Steps to reproduce
How can we trigger this problem?
Try to OTA a 1MB device

Does the problem persist after powering off and on? (just resetting isn't enough sometimes)
yes

### Expected behavior
Tell us what should happen?
Uploading a new version with same major should work

### Actual behavior
Tell us what happens instead?
I had the v2.0.0-dev7 build on a Sonoff S20 and uploaded the dev8 (both from the git) and the upload worked but then I got a simple "flash failed" on the web page. Even worse nothing was working any more. Trying a hard reset (RX<->TX) didn't work either. I had to serial flash and hard reset.

### System configuration
Hardware:
Sonoff S20 (unmodified)

Software or git version:
v2.0.0-dev8

krikk, 2017-05-17:
is most likely the problem that the current build is too large for 1m devices like the sonoff s20, if you compile yourself and remove a few plugins to bring size down it should work

psy0rz, 2017-05-18:
yes it probably is. i think there is some kind of check that hides the OTA option if the flash size is too small, but maybe we have to change it a bit.

indeed compiling it yourself with only the plugins you need should help as well.

mvdbro, 2017-05-21:
If larger sketches are needed, using OTA on a 1MB module, you could use a "two stage upload trick".
I've used this many times with success in the past. I made the smallest thinkable sketch (approx 228kb size) that only loads the config from sketch, connects to wifi and starts the webserver with the httpupdater loaded.

So you first upload the small "uploader" firmware and then upload the new ESP Easy firmware.

Max OTA upload size comparison:
When using ESP Easy to upload on modules with 1MB flash       : 468k
Using this uploader firmware prior to uploading a new ESP Easy: 600k

CherAlban, 2017-06-02:
That sounds at least like a feasible workaround. Could you make the "uploader" firmware available?

psy0rz, 2017-06-05:
isn't the uploader just ESPEasy without any plugins?

psy0rz, 2017-06-05:
perhaps we should make this more clear on the OTA page for 1Mb devices.

CherAlban, 2017-06-06:
What about making the binary available, as everybody with 1MB devices will need it?

krikk, 2017-06-06:
@CherAlban : look in the [wiki](https://www.letscontrolit.com/wiki/index.php/ESPEasy#2.0.2B_firmware_on_1MB_modules)

CherAlban, 2017-06-06:
Excellent! Thanks!

psy0rz, 2017-06-10:
nice thanks:)

closed 2017-06-10 by psy0rz

CherAlban, 2018-01-03:
I am using the ESPEasyUploaderMega to update my Sonoff S20 1MB devices. It works well on older devices, but it fails to update newer devices that need DOUT flashing. Am I doing something wrong?

CherAlban, 2018-01-05:
I tested again, it successfully installs the ESPEasyUploaderMega, but when upgrading to dev13, the web-interface never comes back. Does the ESPEasyUploaderMega image need to have DOUT flashing enabled when compiling?
I also have a problem with a Sonoff basic, which does not even accept the ESPEasyUploaderMega image.

Is somebody still reading this closed issue, or should I open a new one?

TD-er, 2018-01-05:
From what version did you do the upgrade to dev13?
The configuration structure is incompatible between <2.0 and 2.0 and newer.

CherAlban, 2018-01-05:
I upgraded from dev12 to dev13, and also tried dev13 to 20180104

CherAlban, 2018-01-05:
I read your comment in [474](https://github.com/letscontrolit/ESPEasy/issues/474) about maybe enabling DOUT in the firmware. Is this true?

psy0rz, 2018-01-05:
we're reading all the new messages, even on closed issues :)

CherAlban, 2018-01-11:
I don't know who prepared the current ESPEasyUploaderMega firmware, I assume this has been done by mvdbro?
Could somebody please comment on whether DOUT hast to be enabled inside the small uploader firmware? And if so, maybe create a version with DOUT enabled?

Superpiffer, 2018-02-15:
I have the same problem, trying to update via OTA a Wemos D1 Mini Lite (ESP8252 based with 1MB). I successfully installed ESPEasyUploaderMega, it says " Flash Size: | 1024 kB Sketch Max Size: | 604 kB" but after the update from ESP_Easy_mega-20180212_normal_ESP8285_1024.bin to ESP_Easy_mega-20180214_normal_ESP8285_1024.bin (573Kb) the device doesn't boot anymore.

TD-er, 2018-02-15:
Could it be that the config  (SPIFF) was partly overwritten and thus the settings got corrupted?

TD-er, 2018-02-15:
I thought the checksum calculation was only informational and not blocking.
There is a checksum error, but since you upgraded from a version without checksum calculation, it is correct there is a checksum error.
So that won't tell us much.

But still, the image size of both firmware versions (intermediate and new version) + the size of SPIFF of both may not exceed the 1M size. And I think it did.
SPIFF on the 1M versions is 256k?

Superpiffer, 2018-02-15:
There's the problem, the ESP8252 has 1M with no SPIFF so maybe ESPEasy OTA is impossibile with this board. Is that right?

TD-er, 2018-02-15:
I always thought SPIFF was just a setting of the image, not the device.
But there may be some limit of 512k addressable at once with the ESP8285? (not sure in what context I read it, but that idea pops up in my head right now)

CherAlban, 2018-02-15:
My S20s are ESP8266 based, so it cannot be an "ESP8285 only" problem.

TD-er, 2018-02-23:
The "competitor" (Tasmota) uses a lightweight in-between firmware to do OTA updates on the 1MB versions.
This inbetween version is about 300k in size.
Tasmota doesn't use SPIFF, so he doesn't have to worry about those settings getting lost during OTA update. (the storage area used by Tasmota is smaller)

I guess OTA on 1M flash is doomed to fail as long as we don't use a two-step flash procedure.

Superpiffer, 2018-02-23:
The problem is that even with a two step procedure, with ESPEasyUploaderMega_1024.bin (as the [wiki](https://www.letscontrolit.com/wiki/index.php/ESPEasy#2.0.2B_firmware_on_1MB_modules) says), my board gets corrupted. I don't have another esp with 1M so I can't test if it works with others devices (i.e. non ESP8285). 
Anyway, i just tested the "competition" and OTA works as expected, even without flashing an intermediate firmware! I like the flexibility of ESPEasy but I don't know if right now it's reliable enough for my pourpose...

TD-er, 2018-02-23:
So this is a proper usecase where it is needed to make smaller versions. Perhaps even custom built, by editing some kind of 'requirements' file?
Then you should be able to strip out things you don't need. For example no need to include OLED library and fonts for a device without OLED.

What 'devices' do you need on a Sonoff? Probably not much more than a switch i guess?

Superpiffer, 2018-02-23:
Yes, in my case I have a Sonoff-like setup (I only need 2 relays and MQTT) so a "stripped" version is probably a better idea. Sooner or later I want to try and compile myself with only needed libraries and try OTA again.

soif, 2018-02-25:
@hphde  @CherAlban  @Superpiffer

Try my brand new EspBuddy at https://github.com/soif/EspBuddy

Once properly configured, it does the 2 steps OTA update in one simple command:
`espbuddy.php upload my_device -v`

You can also simply fetch (from the firmwares/ directory)  a pre-compiled small OTA firmware that you can upload thru the ESPEasy web interface, and do the second step also via the web.

I've made dozen of OTA tests on various devices (nodemcu, wemos) and it works great. I've made less tests on 1M devices (sonoff TH16, ESP-01) but it should also work. BTW while testing the sonoff TH16 (1M), i had to have -DOUT on serial , and be sure to use the **LATEST development branch version** (with the PUYA fixes included), else the device would hang on boot

_Please report success related to ESPEasy here, while report issues with EspBuddy related stuff in her own repo_

hphde, 2018-02-25:
Thanks @soif but I soldered 4MB on all my 1MB devices

stikla, 2018-03-10:
ESPEasyUploader.OTA.1m128.esp8266.bin works with Sonoff basic flashed with ESPEasy v2.0.0-dev12 and ESPEasy v2.0.0-dev13.

ESPEasyUploader.OTA.1m128.esp8266.bin can be used as replacement for ESPEasyUploaderMega as it does not work with DOUT devices.

soif, 2018-03-10:
>@soif ESPEasyUploader.OTA.1m128.esp8266.bin (from the /firmwares directory) works with >Sonoff basic originally flashed with ESPEasy v2.0.0-dev12 and ESPEasy v2.0.0-dev13.

GOOD news! Thanks for reporting it. 👍 

Would you (or anyone having tested this firmware) pleae add your own tested hardware at:
https://github.com/soif/EspBuddy/wiki/Tested-Hardware

stikla, 2018-03-10:
I must correct myself...

The (only) successful configurations was:
- over serial: erase chip
- over serial: flash ESPEasy v2.0.0-dev13
- over OTA: ESPEasyUploader.OTA.1m128.esp8266.bin
- over OTA: ESPEasy v2.0.0-dev13

But today nothing works... at least not every time.

stikla, 2018-03-12:
For those interested [here](http://applauz.de/?p=4) you can download DOUT version of known ESP Easy Uploader for 1M devices  - ESPEasyUploaderMega_1024.bin

The bin file size is same as original - 270 KB (276.848 bytes)