letscontrolit/ESPEasy #305: Unable to connect to open SSID

Nornode, 2017-05-19:
### Steps to reproduce
Fresh ESP with 2.0.0 dev8, Get SSID scan result and choose the only result (my open wifi).
No input in the password field and setup process can't proceed.

### Expected behaviour
No input in Password field should and would mean it's an open SSID without encryption, right?

### Actual behaviour
It stops from proceeding the setup process.
No log output in serial monitor.

### System configuration
D1 Mini, Chrome and Explorer (edge) tested on a Windows 10 machine.

Software or git version:
2.0.0 dev8 4Mb normal version Pre-compiled .bin

Nornode, 2017-05-24:
Is this a known limitation for ESPeasy?

psy0rz, 2017-05-26:
its probably a bug

sassod, 2017-05-26:
I have similar issue with hidden/not brodcasted SSID by the router....

closed 2017-06-10 by psy0rz