letscontrolit/ESPEasy #307: AM2320 does not report negative temperature values (bellow zero Celsius)

nikil511, 2017-05-21:
### Steps to reproduce
Put an AM2320 in the freezer

### Expected behavior
Show values bellow zero

### Actual behavior
values drop till zero and start going up again 

### System configuration

Hardware:
wemos d1 mini + BME280 + Si7021 + AM2320 on batteries

Software or git version:
build from source mega branch 22/5/2017

krikk, 2017-05-22:
did not test this, but this seems to be a problem in the used library, will have a look when i find time...

...just out of interesst, why do you use an AM2320 when you also have a bme280/si7021?

beicnet, 2017-05-22:
Why not?! I have BME280, HTU21D, LM75A and AM2320 on the same unit...(testing, comparing purposes)!

nikil511, 2017-05-22:
I am building a small 3d printed gadget with a few sensors. Noticed that
readings are off probably due to esp heat in the enclosure but since i have
loads of sensor variations decided to evaluate their performance anyway
just to be sure.

ESPeasy makes these tests so easy 😉

psy0rz, 2017-05-22:
is it possible we use an unsigned variable somewhere?

psy0rz, 2017-05-27:
krikk fixed it

closed 2017-05-27 by psy0rz