letscontrolit/ESPEasy #309: Escaped commands in tools page

jkDesignDE, 2017-05-24:
### Steps to reproduce
Enter a command with params (e.g. `"ABC, 1,2,3"`) in tools page

### Expected behavior
Commands should be send to command interpreter as is

### Actual behavior
If a command has spaces or commas, the characters are escaped (`","` -> `"%44"`, `" "` -> `"+"`). With this extras the command is unknown

### System configuration
Any

psy0rz, 2017-06-05:
ah this needs some html entities decoding. I think I did something similar for the rules page a while ago.

mvdbro, 2017-06-17:
Can't reproduce this issue. Looks like URLDecode/Encode failing. We had this at the very start of the project when the core library failed to handle this properly. If you get stuff like spaces being replace with "+" there must be a bug in the Arduino core. Are you using 2.3.0?

jkDesignDE, 2017-06-17:
I tested it again with actual PlatformIO and actual sources - it works?!?

closed 2017-06-17 by jkDesignDE