letscontrolit/ESPEasy #311: INA219 not working with Domoticz on Wemos D1 mini

MosselNL, 2017-05-25:
### Steps to reproduce
Connect INA219 and configure in ESPEasy with Domoticz as controller. (HTTP or MQTT)

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes

### Expected behavior
It should send his values to Domoticz.

### Actual behavior
No values of the INA219 sensor are been send to Domoticz, i've tried it with HTTP and MQTT. Other sensors have the option "Send Data to controller 1" wich INA219 doesn't have?

### System configuration
Hardware:
Wemos D1 Mini 

Software or git version:
Release v2.0.0-dev8 (normal)

krikk, 2017-05-30:
i will have a look at this...

mvdbro, 2017-06-10:
The plugin was designed to send a single value per task as Domoticz has no custom triple value devices.
Is it  verified to still work with Domoticz?

psy0rz, 2017-06-12:
it wasn't , but someone fixed it, so it should be selectable which single value is send.

closed 2017-06-12 by psy0rz