letscontrolit/ESPEasy #313: P013_HCSR04.ino wrong sensor type and rounding

Grovkillen, 2017-05-27:
### Steps to reproduce
How can we trigger this problem? Try the plugin.

### Expected behavior
I'd expect to be able to use the ultrasonic sensor as a measuring device which lets me round the values using the standard "Decimals" setting. 

### Actual behavior
I am not able to use sub-centimeter (which is actually presented in the log). FormulaOption need to be set to "true" as well. I might want to measure other than the true echo distance with a reference distance (i.e. if the sensor is in the sealing and I want to measure the my own height I might want to have the formula set as, for example, 235-%value%. 235 would be the height from the sealing to the floor.)
The GPIOs in the webform need to be named "Trig" and "Echo" for easier understanding (standard way of presenting plugins).

Extra things to fix:
Row 107 will place double string in the log file. [Picture](http://i.imgur.com/geKolFf.png)
Even though the log file tells  me that the reading is not correct the device works as intended. [Picture](http://i.imgur.com/WGjATpc.png)

### System configuration
2.0.0-dev7

### Hardware:
Wemos D1 mini and NodeMCU v3 + MCSR04

Grovkillen, 2017-05-27:
It would be great to have the plugin able to run PLUGIN_TEN_PER_SECOND even for value since I'd like to use it to measure things that pass underneath it. (in a doorway for example). Could be optional and presented with a warning in the webform (if it is not recommended).

psy0rz, 2017-06-05:
I happen to have this sensor, so I will add it to The Rig as one of the first ones. :)

mvdbro, 2017-06-10:
If you use state mode, it runs PLUGIN_TEN_PER_SECOND...

Grovkillen, 2017-06-10:
Good information. Still value is only cm right?

We need to do a complete page for this sensor since it is not documented well. Also, it is listed under "Ultrasonic sensor", this should be changed to "Distance sensor" since that is more generic. We don't want multiple distance variants. Laser distance sensor for example, or even the pull rope distance sensor could be used in the future.

psy0rz, 2017-06-12:
probably fixed in that last commit referenced above. if its not reopen it :)

closed 2017-06-12 by psy0rz