letscontrolit/ESPEasy #316: Missing Supported Hardware on Wiki

beicnet, 2017-05-28:
As I see on the Wiki page, there are missing a bunch of devices from "Supported Hardware" what I'm using:

1. ESP-01
2. ESP-1
3. ESP-07
4. NodeMCU v0.9
5. NodeMCU v1.0
6. D1 mini Lite
7. WeMos D1 R1 ("Retired")
8. ESPDuino (WROOM-02 aka ESP-13)

Regards

Grovkillen, 2017-05-28:
Please add links to more info. That would help us tremendously.

ShardanX, 2017-06-20:
I'm up to write a wiki posting about the ESP-types with memory size etc,

It will not be complete as i don't have all types in my shelf but
it will be a nice point for starting.

Some ESP versions are a bit outdated:
Old ESP-01, ESP-07, ESP 201: only 512KB flash, it should be mentioned
these will run into problems with ESPEasy V2.0.0 due to the size of the compiled sketch.

Should we mention if they have a FCC as this is the label that allows using a
transmitting device in some countries?

Regards
Shardan

Grovkillen, 2017-06-20:
Yes Shardan, sounds like a really good start.

Grovkillen, 2017-06-20:
We should mention the old ones as well. Just for the reasons you mentioned.

beicnet, 2017-06-20:
@ShardanX Wrong, ESP-01 (black), ESP-01S, ESP-1 (tin covered) and ESP8285 are with 1MB of flash.

ShardanX, 2017-06-20:
That's why i wrote "Old ESP-01". The blue 512 MB version is nearly out of market.
ESP-01S, ESP-1 and ESP8255 are those i haven't in my shelf :)
Only 8255 embedded in Sonoff basic is availlable.

beicnet, 2017-06-20:
@ShardanX Maybe I missed that, sorry, can be mentioned, but abnormal for current releases...
Anyway, I have them all, in all possible variations...just order them from Ali...

Grovkillen, 2018-01-17:
https://www.letscontrolit.com/wiki/index.php/ESP-01

What else can we tick of this list?

TD-er, 2018-01-17:
Perhaps you could add a reference to the "new" ones with the flash chips labeled "PUYA".
These currently have issues flashing from within ESPeasy (and other ESP projects). See #650 and https://github.com/esp8266/Arduino/issues/4061#issuecomment-357810624

closed 2023-09-03 by TD-er