letscontrolit/ESPEasy #317: Enhancement more commands

Grovkillen, 2017-05-29:
We need more commands to set 
Unit name
Unit number
Admin password
Fallback SSID
Fallback WPA key

https://letscontrolit.com/wiki/index.php/ESPEasy_Command_Reference

See discussion here:
https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=3131

Grovkillen, 2017-06-05:
I'm starting to write my own WrapperGUI for the esptool.exe with post-flashing writing using the serial commands. Would it be possible to also have the rules written through the serial port or is to much to ask for?

psy0rz, 2017-06-05:
have to look into this. couldnt your tool just configure the wifi, ping until its online and then upload rules via http?

Grovkillen, 2017-06-05:
@psy0rz yeah it is possible. I will do that. Thanks for looking into it.

psy0rz, 2017-06-05:
still I will add the commands mentioned above, since that's pretty easy and can be usefull.

Grovkillen, 2017-06-16:
Will this not be considered to be of 2.0.0 milestone level?

psy0rz, 2017-06-21:
it is a feature enhancement and we want to focus on fixing bugs, not possibly introducing new ones. but because its you I can make an exception. (should be just copy/past and add some simple commands)

Grovkillen, 2017-06-21:
I'm flattered!!! 😎😂👍

psy0rz, 2017-07-16:
@Grovkillen  i implemented it, ssid2 and key2 where already implemented (see Command.ino for details) 

could you update the commandlist on the wiki?

closed 2017-07-16 by psy0rz

Grovkillen, 2017-07-16:
@psyr0z will fix!

Grovkillen, 2017-07-16:
( @psy0rz I tag you wrong.)

Grovkillen, 2017-07-16:
@psy0rz this is now live...
https://www.letscontrolit.com/wiki/index.php/ESPEasy_Command_Reference