letscontrolit/ESPEasy #318: Inverted Switch State not applied until next switch action.

Grovkillen, 2017-05-29:
When I apply Inverted State the variable is not changed. This only happens upon next switch action. I'd expect the setting to be applied to the variable (i.e. 1-->0 or 0-->1) directly. It is confusing otherwise.

2.0.0-dev9

psy0rz, 2017-06-05:
its understandable that it's implemented this way but indeed confusing. will fix it.

jkDesignDE, 2017-06-09:
It is difficult to solve without affecting the push button state. This conflicts also with an unchecked "Send Boot state".
The only solution (I can think off) would be to send switch value on every "submit" in web interface.

Work around: on changing the "inversed" checkbox check also the "Send Boot state" and press "submit". Then uncheck the  "Send Boot state" and press "submit" again.

Grovkillen, 2017-06-09:
Okay, I understand the technical side of the matter. If not a good solution can be implemented we should state the fact about this in a comment. 

"The switch state will be updated when the next switch action happens" or something like that. As of now you need to know the behind-the-scene to know why the value is not changed when I check the inverted state and hit submit.

The Send Boot State workaround, can't that be used when submitting the inverted state?

jkDesignDE, 2017-06-09:
The Automatically-send is the problem! If the home-controller (e.g. OpenHab) uses the tranmission as is, you can send as many times you want. If the home-controller ises the call to toggle the state of an actuator, it goes on and off on each transmission. I think that was the idea behind the checkbox "Send Boot state".

If you (and all others) accept that the actual value is send on every web-page-submit (ignoring "Send Boot state"), there is an easy solution.

Grovkillen, 2017-06-09:
As long as we explain the behavior I'm fine with it. I understand the technical side now. Lets add some small explanation into the WebGUI?

commit 29627100 (2017-07-16): added note to inverse logic to fix #318
closed 2017-07-16 by psy0rz